intel/media-driver #290: [ffmpeg-vaapi][vpp_sharpness] the vpp_sharpness failed on CFL/KBL/SKL platforms by using ffmpeg-vaapi with iHD driver

zhuqingliang, 2018-08-27:
Test Env:
==================================
platform: CFL/KBL/SKL
Arch: x86_64
Kernel version: CFL 4.15.0,    KBL  4.14.0-rc1 , SKL  4.15.0
Libva: 6ae717304c48e7710a76a87cef2a1096ff450af2
media_driver: 36b3a0a
libva-utils:  9a10ad663349732decd668a426cfc349b0010d1d
ffmpeg-vaapi: 5423fe29da6bbf28f5253800cace133ac8343a7e(master branch)

Reproduce Steps:
======================================
1.   build enc as above lists
2.   ffmpeg -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -f rawvideo -s:v 4096x2304 -i /media/ffmpeg_stream_dcg/SecretWorld_4096x2304_500Frames.yuv -vf format=nv12,hwupload,sharpness_vaapi=sharpness=10,hwdownload,format=nv12 -an -vframes 20 -f rawvideo -y vpp_4K_sharpness_10.yuv 
3.  vpp_4K_sharpness_10.yuv is all black  image.

Failed test case:
=============================
vpp_4K_sharpness_10.yuv
vpp_4K_sharpness_30.yuv
vpp_4K_sharpness_60.yuv

FurongZhang, 2018-09-04:
I will check that and update the status next week.

Jister, 2018-09-27:
@zhuqingliang Since we could not find 4096x2304 content, we use 4096x2160 content, and it can generate right output. And could you please share your example content "SecretWorld_4096x2304_500Frames.yuv" ?

zhuqingliang, 2018-09-30:
@Jister  I open the origin file  SecretWorld_4096x2304_500Frames.yuv  using  video quality caliper , there is 30+ frames is black . This issue is invaild . @

closed 2018-09-30 by zhuqingliang