intel/media-driver #291: [FFmpeg-Vaapi][vpp] hevc transcoding fail from other format on CFL/KBL with iHD driver

zhuqingliang, 2018-08-28:
System Environment
====================
Platform: CFL/KBL
Arch: x86_64
Kernel version: CFL 4.15.0, KBL 4.14.0-rc1
Libva: 6ae717304c48e7710a76a87cef2a1096ff450af2
media_driver: 36b3a0aa3b47a4cda36a31d98e16da9ff7ea27fb
libva-utils: 9a10ad663349732decd668a426cfc349b0010d1d
ffmpeg-vaapi: 5423fe29da6bbf28f5253800cace133ac8343a7e(master branch)

Bug detailed description
=======
ffmpeg transcoding from h264 to hevc error

Reproduce Steps
==============
1. build enc as above lists
2. ffmpeg -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -v verbose -hwaccel_output_format vaapi -i /media//ffmpeg_stream_dcg//4K_Time_Lapse_Footage-American_Cities.h264 -vf scale_vaapi=w=1920:h=1080 -c:v hevc_vaapi -vframes 50 -an -y ./vpp_scaling_h264_hevc_4096x2304_1920x1080.hevc
3. 
[hevc_vaapi @ 0xef8e00] Failed to end picture encode issue: 24 (internal encoding error).
[hevc_vaapi @ 0xef8e00] Encode failed: -5.
Video encoding failed

zhuqingliang, 2018-08-29:
This issue  _don't_ exist on i965 driver .

XinfengZhang, 2018-09-03:
please use the commit  after 5c8a663

closed 2018-10-16 by zhuqingliang