intel/media-driver #294: CpuSwizzleBlt.c: No such file or directory

Brainiarc7, 2018-08-30:
So, building the latest iteration of this driver (current git tip, master branch) results in this:

```
In file included from /usr/include/igdgmm/inc/GmmLib.h:48:0,
                 from /home/bob/vaapi/workspace/media-driver/media_driver/linux/common/os/linux_shadow_skuwa.h:31,
                 from /home/bob/vaapi/workspace/media-driver/media_driver/linux/common/os/media_skuwa_specific.h:32,
                 from /home/bob/vaapi/workspace/media-driver/media_driver/agnostic/common/os/mos_os.h:31,
                 from /home/bob/vaapi/workspace/media-driver/media_driver/agnostic/common/cm/cm_hal.cpp:26:
/usr/include/igdgmm/inc/External/Common/GmmCommonExt.h:60:10: fatal error: ../../../Utility/CpuSwizzleBlt/CpuSwizzleBlt.c: No such file or directory
 #include "../../../Utility/CpuSwizzleBlt/CpuSwizzleBlt.c"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/build.make:230: recipe for target 'media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/agnostic/common/cm/cm_hal.cpp.o' failed
make[2]: *** [media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/agnostic/common/cm/cm_hal.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /usr/include/igdgmm/inc/GmmLib.h:48:0,
                 from /home/bob/vaapi/workspace/media-driver/media_driver/linux/common/os/linux_shadow_skuwa.h:31,
                 from /home/bob/vaapi/workspace/media-driver/media_driver/linux/common/os/media_skuwa_specific.h:32,
                 from /home/bob/vaapi/workspace/media-driver/media_driver/agnostic/common/os/mos_os.h:31,
                 from /home/bob/vaapi/workspace/media-driver/media_driver/agnostic/common/cm/cm_hal_hashtable.h:29,
                 from /home/bob/vaapi/workspace/media-driver/media_driver/agnostic/common/cm/cm_hal_hashtable.cpp:36:
/usr/include/igdgmm/inc/External/Common/GmmCommonExt.h:60:10: fatal error: ../../../Utility/CpuSwizzleBlt/CpuSwizzleBlt.c: No such file or directory
 #include "../../../Utility/CpuSwizzleBlt/CpuSwizzleBlt.c"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/build.make:254: recipe for target 'media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/agnostic/common/cm/cm_hal_hashtable.cpp.o' failed
make[2]: *** [media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/agnostic/common/cm/cm_hal_hashtable.cpp.o] Error 1
In file included from /usr/include/igdgmm/inc/GmmLib.h:48:0,
                 from /home/bob/vaapi/workspace/media-driver/media_driver/linux/common/os/linux_shadow_skuwa.h:31,
                 from /home/bob/vaapi/workspace/media-driver/media_driver/linux/common/os/media_skuwa_specific.h:32,
                 from /home/bob/vaapi/workspace/media-driver/media_driver/linux/common/os/mos_os_specific.h:30,
                 from /home/bob/vaapi/workspace/media-driver/media_driver/linux/common/os/libdrm/include/mos_bufmgr.h:55,
                 from /home/bob/vaapi/workspace/media-driver/media_driver/linux/common/ddi/media_libva_common.h:36,
                 from /home/bob/vaapi/workspace/media-driver/media_driver/linux/common/cm/media_libva_cm.h:30,
                 from /home/bob/vaapi/workspace/media-driver/media_driver/linux/common/cm/cm_def_os.h:41,
                 from /home/bob/vaapi/workspace/media-driver/media_driver/agnostic/common/cm/cm_def.h:30,
                 from /home/bob/vaapi/workspace/media-driver/media_driver/agnostic/common/cm/cm_def.cpp:23:
/usr/include/igdgmm/inc/External/Common/GmmCommonExt.h:60:10: fatal error: ../../../Utility/CpuSwizzleBlt/CpuSwizzleBlt.c: No such file or directory
 #include "../../../Utility/CpuSwizzleBlt/CpuSwizzleBlt.c"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/build.make:134: recipe for target 'media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/agnostic/common/cm/cm_def.cpp.o' failed
make[2]: *** [media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/agnostic/common/cm/cm_def.cpp.o] Error 1
In file included from /usr/include/igdgmm/inc/GmmLib.h:48:0,
                 from /home/bob/vaapi/workspace/media-driver/media_driver/linux/common/os/linux_shadow_skuwa.h:31,
                 from /home/bob/vaapi/workspace/media-driver/media_driver/linux/common/os/media_skuwa_specific.h:32,
                 from /home/bob/vaapi/workspace/media-driver/media_driver/linux/common/os/mos_os_specific.h:30,
                 from /home/bob/vaapi/workspace/media-driver/media_driver/linux/common/os/libdrm/include/mos_bufmgr.h:55,
                 from /home/bob/vaapi/workspace/media-driver/media_driver/linux/common/ddi/media_libva_common.h:36,
                 from /home/bob/vaapi/workspace/media-driver/media_driver/linux/common/cm/media_libva_cm.h:30,
                 from /home/bob/vaapi/workspace/media-driver/media_driver/linux/common/cm/cm_def_os.h:41,
                 from /home/bob/vaapi/workspace/media-driver/media_driver/agnostic/common/cm/cm_def.h:30,
                 from /home/bob/vaapi/workspace/media-driver/media_driver/agnostic/common/cm/cm_buffer.h:30,
                 from /home/bob/vaapi/workspace/media-driver/media_driver/agnostic/common/cm/cm_buffer_rt.h:30,
                 from /home/bob/vaapi/workspace/media-driver/media_driver/agnostic/common/cm/cm_state_buffer.h:30,
                 from /home/bob/vaapi/workspace/media-driver/media_driver/agnostic/common/cm/cm_state_buffer.cpp:27:
/usr/include/igdgmm/inc/External/Common/GmmCommonExt.h:60:10: fatal error: ../../../Utility/CpuSwizzleBlt/CpuSwizzleBlt.c: No such file or directory
 #include "../../../Utility/CpuSwizzleBlt/CpuSwizzleBlt.c"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from /usr/include/igdgmm/inc/GmmLib.h:48:0,
                 from /home/bob/vaapi/workspace/media-driver/media_driver/linux/common/os/linux_shadow_skuwa.h:31,
                 from /home/bob/vaapi/workspace/media-driver/media_driver/linux/common/os/media_skuwa_specific.h:32,
                 from /home/bob/vaapi/workspace/media-driver/media_driver/linux/common/os/mos_os_specific.h:30,
                 from /home/bob/vaapi/workspace/media-driver/media_driver/linux/common/os/libdrm/include/mos_bufmgr.h:55,
                 from /home/bob/vaapi/workspace/media-driver/media_driver/linux/common/ddi/media_libva_common.h:36,
                 from /home/bob/vaapi/workspace/media-driver/media_driver/linux/common/cm/media_libva_cm.h:30,
                 from /home/bob/vaapi/workspace/media-driver/media_driver/linux/common/cm/cm_def_os.h:41,
                 from /home/bob/vaapi/workspace/media-driver/media_driver/agnostic/common/cm/cm_def.h:30,
                 from /home/bob/vaapi/workspace/media-driver/media_driver/linux/common/cm/cm_device.h:30,
[... 83 lines cut ...]
make[1]: *** Waiting for unfinished jobs....
[  6%] Linking CXX shared library libdrm_mock.so
cd /home/bob/vaapi/workspace/build_media/media_driver/linux/ult/libdrm_mock && /usr/bin/cmake -E cmake_link_script CMakeFiles/drm_mock.dir/link.txt --verbose=1
/usr/bin/c++ -fPIC  -Wno-reorder -Wsign-promo -Wnon-virtual-dtor -Wno-invalid-offsetof -fvisibility-inlines-hidden -fno-use-cxa-atexit -frtti -fexceptions -fpermissive -fcheck-new -std=c++1y -std=c++11  -shared -Wl,-soname,libdrm_mock.so.2 -o libdrm_mock.so.2.4.0 CMakeFiles/drm_mock.dir/mos_bufmgr_api_mock.c.o CMakeFiles/drm_mock.dir/mos_bufmgr_mock.c.o CMakeFiles/drm_mock.dir/xf86drmHash_mock.c.o CMakeFiles/drm_mock.dir/xf86drmMode_mock.c.o CMakeFiles/drm_mock.dir/xf86drmRandom_mock.c.o CMakeFiles/drm_mock.dir/xf86drm_mock.c.o -lpciaccess 
cd /home/bob/vaapi/workspace/build_media/media_driver/linux/ult/libdrm_mock && /usr/bin/cmake -E cmake_symlink_library libdrm_mock.so.2.4.0 libdrm_mock.so.2 libdrm_mock.so
make[2]: Leaving directory '/home/bob/vaapi/workspace/build_media'
[  6%] Built target drm_mock
[  7%] Linking CXX static library liblibgtest.a
cd /home/bob/vaapi/workspace/build_media/media_driver/linux/ult/ult_app/googletest && /usr/bin/cmake -P CMakeFiles/libgtest.dir/cmake_clean_target.cmake
cd /home/bob/vaapi/workspace/build_media/media_driver/linux/ult/ult_app/googletest && /usr/bin/cmake -E cmake_link_script CMakeFiles/libgtest.dir/link.txt --verbose=1
/usr/bin/ar qc liblibgtest.a  CMakeFiles/libgtest.dir/src/gtest-all.cc.o CMakeFiles/libgtest.dir/src/gtest_main.cc.o
/usr/bin/ranlib liblibgtest.a
make[2]: Leaving directory '/home/bob/vaapi/workspace/build_media'
[  7%] Built target libgtest
make[1]: Leaving directory '/home/bob/vaapi/workspace/build_media'
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2

real	0m5.213s
user	0m10.114s
sys	0m1.258s

```

**Platform details:**

mypopydev, 2018-08-30:
@Brainiarc7 I resolved this issue after deleted the old gmmlib in /usr/local/include/igdgmm/, you can try this way.

yangnin, 2018-09-03:
@Brainiarc7 Do you have tried for this?  Media driver has swtiched to dynamic gmm lib.

Brainiarc7, 2018-09-07:
Yes, the issue is resolved in the latest driver. Closing this now.

closed 2018-09-07 by Brainiarc7