intel/media-driver #295: [FFmpeg-Vaapi][HEVC_ffmpeg_decode]the dumped file has garbage on CFL on iHD driver

zhuqingliang, 2018-08-30:
System Environment
=======
Platform: CFL
Arch: x86_64
Kernel version: CFL 4.15.0, 
Libva: 6ae717304c48e7710a76a87cef2a1096ff450af2
media_driver: 36b3a0aa3b47a4cda36a31d98e16da9ff7ea27fb
libva-utils: 9a10ad663349732decd668a426cfc349b0010d1d
ffmpeg-vaapi: 5423fe29da6bbf28f5253800cace133ac8343a7e(master branch)

Regression?
=======

Bug detailed description
=======
the dumped file has garbage with iHD driver, and the md5 value does not match with the reference on CFL

Reproduce Steps
==============
1.ffmpeg -hwaccel vaapi -hwaccel_device /dev/dri/renderD128 -v verbose -y -i /media//ffmpeg_hevc/AMP_A_Samsung_6.bit -pix_fmt yuv420p -y ./AMP_A_Samsung_6.bit.yuv
2. real_md5=a24c19048a55394de8c1b454f450888d ref_md5=86afaedd98cbc5f11da775121bf12dd5
3. use tool 7yuv to observe the dumped file

Expected Result
=============
the dumped file has no garbage and the md5 match with the reference

Actual Result
===========
the dumped file has garbage

Failed test case
=================
AMP_A_Samsung_6.bit
AMP_B_Samsung_6.bit
AMVP_C_Samsung_6.bit
STRUCT_B_Samsung_6.bit

XinfengZhang, 2018-09-03:
@zhuqingliang where can get the stream? is it a public stream?

XinfengZhang, 2018-10-11:
cant reproduce this issue with latest media driver. commit 60b1e83aef1b751fb2b455b67766dc579cd73c80

closed 2018-10-16 by zhuqingliang