intel/media-driver #299: Build error: CMake Error at media_driver/media_top_cmake.cmake:92 (add_subdirectory)

dbermond, 2018-09-02:
I'm getting the following error when trying to build intel-media-driver git master:

```
CMake Error at media_driver/media_top_cmake.cmake:92 (add_subdirectory):
  add_subdirectory given source
  "/storage/linux/abs/intel-media-driver-git/src/media-driver/../gmmlib/Source/GmmLib"
  which is not an existing directory.
Call Stack (most recent call first):
  media_driver/CMakeLists.txt:111 (include)

-- LIBVA_DRIVERS_PATH = /usr/lib/dri
-- Configuring incomplete, errors occurred!
```

- A git bisect shows that the offending commit is 1383ce43764ff12c73c3d594cd4a63476883dac9.

- I can confirm that it was building fine up until commit 10f382d1.

- Complete cmake command and output: [https://bpaste.net/show/362d3a855b14](https://bpaste.net/show/362d3a855b14).

- Workspace is setted according to the most recent [README.md](https://github.com/intel/media-driver/blob/master/README.md) file.

- gmmlib is installed on system. The same error is shown if gmmlib is uninstalled.

**System Information:**
**OS:** Arch Linux x86_64
**Compiler:** gcc 8.2.0
**CMake:** 3.12.1
**intel-media-driver:** git master
**gmmlib:** git master
**libva:** git master

yangnin, 2018-09-03:
Please help to review PR#301, it would be OK for your compilation.

https://github.com/intel/media-driver/pull/301

dbermond, 2018-09-03:
@yangnin Thank you for the reply.

PR #301 changes only the file `README.md`. No code is changed on it that can fix this issue. This readme change only instructs to have gmmlib git master installed, which I already have. My installed gmmlib is currently at commit https://github.com/intel/gmmlib/commit/99ff76418179b4fb069235ff6c028330c802563a.

dvrogozh, 2018-09-03:
@yangnin : I strongly suggest to consider the following:
* If you submit anything to the repository, commit message should clearly state what was changed and why
* Avoid usage of non-standard practices and words in the commit message. Current "Backout" process is absolutely not clear from the perspective of good commit message, community standard practices and even naming of the process. IMHO, you have substituted `git-revert`! As a result you can reasonably expect that people will not understand what was done.
* Avoid reverting changes depending on the internal state, i.e. avoid situations when media-driver is not affected in any externally visible way, but you still revert changes because of some internal considerations

dbermond, 2018-09-03:
@dvrogozh Thank you for your very explanatory comment. It makes it clear to understand what's going on with theses changes in the gmmlib requirement.

Now I can build intel-media-driver with the old cmake command that points to local gmmlib sources, although the actual `README.md` does not mention it anymore.

dbermond, 2018-09-08:
The use of dynamic gmmlib is now enabled again in the current git master branch since September 6, 2018 through commit https://github.com/intel/media-driver/commit/63dd4ae121542e281ffd76a943b8ece1f18acbb5. It solves this issue.

I think this issue can be closed.

Thank you.

closed 2018-09-08 by dbermond

dvrogozh, 2018-09-10:
@dbermond : you were ahead of me tor write this down:). Yes, gmmlib change was re-introduced and now media-driver should be able to properly detect it via pkg-config. Eventually, this will be available in the upcoming 18.3 release.