intel/media-driver #308: Build error with stable libva

pryg-skok, 2018-09-07:
I'm trying to build the newest but stable MediaSDK on ubuntu 16.04.4.
Based on wiki I've updated my compiler to gcc-6:
```# gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/6/lto-wrapper
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 6.4.0-17ubuntu1~16.04' --with-bugurl=file:///usr/share/doc/gcc-6/README.Bugs --enable-languages=c,ada,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --with-as=/usr/bin/x86_64-linux-gnu-as --with-ld=/usr/bin/x86_64-linux-gnu-ld --program-suffix=-6 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-libmpx --enable-plugin --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-6-amd64/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-6-amd64 --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-6-amd64 --with-arch-directory=amd64 --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --with-target-system-zlib --enable-objc-gc=auto --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 6.4.0 20180424 (Ubuntu 6.4.0-17ubuntu1~16.04) 
```

Also on wiki I've read that I should use specific libva version:
`git checkout tags/2.1.1.pre1-20180601`

Next I've tried to build master commit of media_driver and it fails with error:
```/root/media-driver/media_driver/linux/common/codec/ddi/media_ddi_decode_hevc.cpp: In member function ‘virtual VAStatus DdiDecodeHEVC::RenderPicture(VADriverContextP, VAContextID, VABufferID*, int32_t)’:
/root/media-driver/media_driver/linux/common/codec/ddi/media_ddi_decode_hevc.cpp:434:14: error: ‘VASubsetsParameterBufferType’ was not declared in this scope
         case VASubsetsParameterBufferType:
              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/build.make:9782: recipe for target 'media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/linux/common/codec/ddi/media_ddi_decode_hevc.cpp.o' failed
make[2]: *** [media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/linux/common/codec/ddi/media_ddi_decode_hevc.cpp.o] Error 1
CMakeFiles/Makefile2:167: recipe for target 'media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/all' failed
make[1]: *** [media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
```

If I try to build the master version of libva there is another error:
```In file included from /root/media-driver/media_driver/linux/common/ddi/media_libva.cpp:45:0:
/root/media-driver/media_driver/linux/common/ddi/media_libva_putsurface_linux.h:31:29: fatal error: va/va_dricommon.h: No such file or directory
 #include <va/va_dricommon.h>
                             ^
compilation terminated.
media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/build.make:10310: recipe for target 'media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/linux/common/ddi/media_libva.cpp.o' failed
make[2]: *** [media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/linux/common/ddi/media_libva.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:167: recipe for target 'media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/all' failed
make[1]: *** [media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
```

So, which commits should I use to successfully media-driver?

pryg-skok, 2018-09-10:
Maybe I need to submit additional info about build environment?

yangnin, 2018-09-14:
Yes, Please give more detail or reproduce steps for your issue.
Thanks!

XinfengZhang, 2018-09-18:
please help to check the version, libva add VASubsetsParameterBufferType recently, and driver also add it, so you need compatible libva version

mort1k, 2018-10-06:
Same error with ubuntu 18:
`-- The C compiler identification is GNU 7.3.0
-- The CXX compiler identification is GNU 7.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- CMAKE_INSTALL_PREFIX = /usr/local
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'libva>=1.0.0'
--   Found libva, version 1.1.0
*BUILD_TYPE not defined, default to: release
-- Checking for module 'igdgmm'
--   Found igdgmm, version 1.0.0
-- media -- PLATFORM = linux
-- media -- ARCH = 64
-- media -- CMAKE_CURRENT_LIST_DIR = /opt/src/media-driver-intel-media-18.3.0/media_driver
-- media -- INCLUDED_LIBS = 
-- media -- LIB_NAME = iHD_drv_video
-- media -- OUTPUT_NAME = 
-- media -- BUILD_TYPE/UFO_BUILD_TYPE/CMAKE_BUILD_TYPE = release/release/Release
-- media -- LIBVA_INSTALL_PATH = 
-- media -- MEDIA_VERSION = 1.0.0
`

`/opt/src/media-driver-intel-media-18.3.0/media_driver/linux/common/codec/ddi/media_ddi_decode_base.cpp: In member function ‘virtual VAStatus DdiMediaDecode::CreateBuffer(VABufferType, uint32_t, uint32_t, void*, VABufferID*)’:
/opt/src/media-driver-intel-media-18.3.0/media_driver/linux/common/codec/ddi/media_ddi_decode_base.cpp:744:14: error: ‘VASubsetsParameterBufferType’ was not declared in this scope
         case VASubsetsParameterBufferType:
              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/opt/src/media-driver-intel-media-18.3.0/media_driver/linux/common/codec/ddi/media_ddi_decode_base.cpp:744:14: note: suggested alternative: ‘VASliceParameterBufferType’
         case VASubsetsParameterBufferType:
              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
              VASliceParameterBufferType
[ 87%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/linux/common/codec/ddi/media_ddi_decode_hevc.cpp.o
media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/build.make:9662: recipe for target 'media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/linux/common/codec/ddi/media_ddi_decode_base.cpp.o' failed
make[2]: *** [media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/linux/common/codec/ddi/media_ddi_decode_base.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/opt/src/media-driver-intel-media-18.3.0/media_driver/linux/common/codec/ddi/media_ddi_decode_hevc.cpp: In member function ‘virtual VAStatus DdiDecodeHEVC::RenderPicture(VADriverContextP, VAContextID, VABufferID*, int32_t)’:
/opt/src/media-driver-intel-media-18.3.0/media_driver/linux/common/codec/ddi/media_ddi_decode_hevc.cpp:434:14: error: ‘VASubsetsParameterBufferType’ was not declared in this scope
         case VASubsetsParameterBufferType:
              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/opt/src/media-driver-intel-media-18.3.0/media_driver/linux/common/codec/ddi/media_ddi_decode_hevc.cpp:434:14: note: suggested alternative: ‘VASliceParameterBufferType’
         case VASubsetsParameterBufferType:
              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
              VASliceParameterBufferType
media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/build.make:9782: recipe for target 'media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/linux/common/codec/ddi/media_ddi_decode_hevc.cpp.o' failed
make[2]: *** [media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/linux/common/codec/ddi/media_ddi_decode_hevc.cpp.o] Error 1
CMakeFiles/Makefile2:238: recipe for target 'media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/all' failed
make[1]: *** [media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2
`

kwende, 2019-04-13:
Is there any further input on this? I'm getting this error too. I'm using the same branch as this guy.

michaelolbrich, 2019-09-10:
`va/va_dricommon.h` is only installed if libva ls build with `-Dwith_x11=yes` / `--enable-x11`. However the header is used here if `X11_FOUND` is defined. And this is defined if CMake found (some?) X11 libs and headers, regardless of the libva configuration.

So the missing include error can happen if the X11 devel packages are installed but libva is compiled with X11 support explicitly disabled.