intel/media-driver #310: [ICL][gstreamer-vaapi][h264/h265 dec] Decode error for fail to map va image.

feiwan1, 2018-09-10:
kernel:4.18.0
gst-vaapi:1.15.0.1

root@wuqiong-Skylake:~/media# LIBVA_TRACE=./vatrace.log gst-launch-1.0 filesrc location=test.h265 ! h265parse ! vaapih265dec ! filesink location=a.yuv
Setting pipeline to PAUSED ...
libEGL warning: DRI2: failed to authenticate
Pipeline is PREROLLING ...
Got context from element 'vaapidecode_h264-0': gst.gl.GLDisplay=context, gst.gl.GLDisplay=(GstGLDisplay)"\(GstGLDisplayX11\)\ gldisplayx11-0";
Got context from element 'vaapidecode_h264-0': gst.vaapi.Display=context, gst.vaapi.Display=(GstVaapiDisplay)"\(GstVaapiDisplayEGL\)\ vaapidisplayegl0";
Redistribute latency...
0:00:00.119604121 11003 0x563143e6fca0 ERROR       vaapivideomemory gstvaapivideomemory.c:852:allocator_configure_image_info: Failed to map VA image 0x7f1764006050
0:00:00.119666981 11003 0x563143e6fca0 ERROR            vaapidecode gstvaapipluginbase.c:608:ensure_srcpad_allocator:<vaapidecode_h264-0> failed to create src pad's allocator
ERROR: from element /GstPipeline:pipeline0/GstH264Parse:h264parse0: Internal data stream error.

Same cmd can work on APL/SKL/KBL. And h264 decode cmd is:
LIBVA_TRACE=./vatrace.log gst-launch-1.0 filesrc location=test.h264 ! h264parse ! vaapih264dec ! filesink location=a.yuv

feiwan1, 2018-09-10:
[vatrace.log](https://github.com/intel/media-driver/files/2365508/vatrace.log)

Tianhaol, 2018-09-11:
Could you tell whether the kernel is installed from source downloaded from kernel.org or is installed  from OS installation?

feiwan1, 2018-09-14:
By using master meida-driver, the issue is gone. Thanks @Tianhaol .

closed 2018-09-14 by feiwan1