intel/media-driver #315: media driver adds out-of-sources gmmlib search paths despite detecting via pkg-config

dvrogozh, 2018-09-13:
One of the media-driver compile commands:
`/usr/lib64/ccache/c++ -DHEVC_FEI_ENABLE_CMRT -DIGFX_GEN10_CNL_SUPPORTED -DIGFX_GEN10_SUPPORTED -DIGFX_GEN11_ICLLP_SUPPORTED -DIGFX_GEN11_SUPPORTED -DIGFX_GEN8_BDW_SUPPORTED -DIGFX_GEN8_SUPPORTED -DIGFX_GEN9_BXT_SUPPORTED -DIGFX_GEN9_CFL_SUPPORTED -DIGFX_GEN9_GLK_SUPPORTED -DIGFX_GEN9_KBL_SUPPORTED -DIGFX_GEN9_SKL_SUPPORTED -DIGFX_GEN9_SUPPORTED -DUFO_VERSION="2.0.0" -DVEBOX_AUTO_DENOISE_SUPPORTED=1 -D_AVC_DECODE_SUPPORTED -D_AVC_ENCODE_VDENC_SUPPORTED -D_AVC_ENCODE_VME_SUPPORTED -D_COMMON_ENCODE_SUPPORTED -D_DECODE_PROCESSING_SUPPORTED -D_HEVC_DECODE_SUPPORTED -D_HEVC_ENCODE_VDENC_SUPPORTED -D_HEVC_ENCODE_VME_SUPPORTED -D_JPEG_DECODE_SUPPORTED -D_JPEG_ENCODE_SUPPORTED -D_MMC_SUPPORTED -D_MPEG2_DECODE_SUPPORTED -D_MPEG2_ENCODE_VME_SUPPORTED -D_RELEASE -D_VC1_DECODE_SUPPORTED -D_VP8_DECODE_SUPPORTED -D_VP8_ENCODE_SUPPORTED -D_VP9_DECODE_SUPPORTED -D_VP9_ENCODE_VDENC_SUPPORTED -D__STDC_CONSTANT_MACROS -D__STDC_LIMIT_MACROS -D__VPHAL_SFC_SUPPORTED=1 -I/include -Igcmedia-driver/media_driver/linux/common/os/libdrm/include -I/include/igdgmm/util -I/include/igdgmm/inc/common -I/include/igdgmm/inc -I/include/igdgmm/GmmLib/inc -I/include/igdgmm/GmmLib -I/include/igdgmm -I/media-driver/media_driver/agnostic/common/cm -I/media-driver/media_driver/agnostic/common/codec/hal -I/media-driver/media_driver/agnostic/common/codec/kernel -I/media-driver/media_driver/agnostic/common/codec/shared -I/media-driver/media_driver/agnostic/common/heap_manager -I/media-driver/media_driver/agnostic/common/hw/vdbox -I/media-driver/media_driver/agnostic/common/hw -I/media-driver/media_driver/agnostic/common/media_interfaces -I/media-driver/media_driver/agnostic/common/os -I/media-driver/media_driver/agnostic/common/renderhal -I/media-driver/media_driver/agnostic/common/shared -I/media-driver/media_driver/agnostic/common/vp/hal -I/media-driver/media_driver/agnostic/common/vp/kdll -I/media-driver/media_driver/agnostic/common/vp/kernel -I/media-driver/media_driver/agnostic/gen8/cm -I/media-driver/media_driver/agnostic/gen8/codec/hal -I/media-driver/media_driver/agnostic/gen8/codec/kernel -I/media-driver/media_driver/agnostic/gen8/hw/vdbox -I/media-driver/media_driver/agnostic/gen8/hw -I/media-driver/media_driver/agnostic/gen8/renderhal -I/media-driver/media_driver/agnostic/gen8/vp/hal -I/media-driver/media_driver/agnostic/gen8/vp/kernel -I/media-driver/media_driver/agnostic/gen8_bdw/codec/hal -I/media-driver/media_driver/agnostic/gen8_bdw/hw/vdbox -I/media-driver/media_driver/agnostic/gen8_bdw/media_interfaces -I/media-driver/media_driver/agnostic/gen8_bdw/renderhal -I/media-driver/media_driver/agnostic/gen9/cm -I/media-driver/media_driver/agnostic/gen9/codec/hal -I/media-driver/media_driver/agnostic/gen9/codec/kernel -I/media-driver/media_driver/agnostic/gen9/hw/vdbox -I/media-driver/media_driver/agnostic/gen9/hw -I/media-driver/media_driver/agnostic/gen9/renderhal -I/media-driver/media_driver/agnostic/gen9/vp/hal -I/media-driver/media_driver/agnostic/gen9/vp/kernel -I/media-driver/media_driver/agnostic/gen9_bxt/codec/hal -I/media-driver/media_driver/agnostic/gen9_bxt/codec/kernel -I/media-driver/media_driver/agnostic/gen9_bxt/hw/vdbox -I/media-driver/media_driver/agnostic/gen9_bxt/media_interfaces -I/media-driver/media_driver/agnostic/gen9_bxt/vp/hal -I/media-driver/media_driver/agnostic/gen9_skl/codec/hal -I/media-driver/media_driver/agnostic/gen9_skl/codec/cmrt_kernel -I/media-driver/media_driver/agnostic/gen9_skl/hw/vdbox -I/media-driver/media_driver/agnostic/gen9_skl/media_interfaces -I/media-driver/media_driver/agnostic/gen9_cfl/media_interfaces -I/media-driver/media_driver/agnostic/gen9_glk/codec/hal -I/media-driver/media_driver/agnostic/gen9_glk/hw/vdbox -I/media-driver/media_driver/agnostic/gen9_glk/media_interfaces -I/media-driver/media_driver/agnostic/gen9_glk/vp/hal -I/media-driver/media_driver/agnostic/gen9_kbl/codec/hal -I/media-driver/media_driver/agnostic/gen9_kbl/codec/kernel -I/media-driver/media_driver/agnostic/gen9_kbl/hw/vdbox -I/media-driver/media_driver/agnostic/gen9_kbl/media_interfaces -I/media-driver/media_driver/agnostic/gen10/cm -I/media-driver/media_driver/agnostic/gen10/codec/hal -I/media-driver/media_driver/agnostic/gen10/codec/kernel -I/media-driver/media_driver/agnostic/gen10/codec/share -I/media-driver/media_driver/agnostic/gen10/hw/vdbox -I/media-driver/media_driver/agnostic/gen10/hw -I/media-driver/media_driver/agnostic/gen10/renderhal -I/media-driver/media_driver/agnostic/gen10/vp/hal -I/media-driver/media_driver/agnostic/gen10/vp/kernel -I/media-driver/media_driver/agnostic/gen10_cnl/media_interfaces -I/media-driver/media_driver/agnostic/gen11/cm -I/media-driver/media_driver/agnostic/gen11/codec/hal -I/media-driver/media_driver/agnostic/gen11/codec/kernel -I/media-driver/media_driver/agnostic/gen11/codec/share -I/media-driver/media_driver/agnostic/gen11/hw/vdbox -I/media-driver/media_driver/agnostic/gen11/hw -I/media-driver/media_driver/agnostic/gen11/renderhal -I/media-driver/media_driver/agnostic/gen11/vp/hal -I/media-driver/media_driver/agnostic/gen11_icllp/codec/hal -I/media-driver/media_driver/agnostic/gen11_icllp/codec/kernel -I/media-driver/media_driver/agnostic/gen11_icllp/media_interfaces -I/media-driver/media_driver/agnostic/gen11_icllp/vp/hal -I/media-driver/media_driver/agnostic/gen11_icllp/vp/kernel -I/media-driver/media_driver/linux/common/cm -I/media-driver/media_driver/linux/common/codec/ddi -I/media-driver/media_driver/linux/common/cp/ddi -I/media-driver/media_driver/linux/common/cp/hw -I/media-driver/media_driver/linux/common/cp/os -I/media-driver/media_driver/linux/common/cp/shared -I/media-driver/media_driver/linux/common/ddi -I/media-driver/media_driver/linux/common/os -I/media-driver/media_driver/linux/common/vp/ddi -I/media-driver/media_driver/linux/gen8/ddi -I/media-driver/media_driver/linux/gen9/ddi -I/media-driver/media_driver/linux/gen9_bxt/ddi -I/media-driver/media_driver/linux/gen9_skl/ddi -I/media-driver/media_driver/linux/gen9_kbl/ddi -I/media-driver/media_driver/linux/gen9_glk/ddi -I/media-driver/media_driver/linux/gen9_cfl/ddi -I/media-driver/media_driver/linux/gen10/ddi -I/media-driver/media_driver/linux/gen10_cnl/ddi -I/media-driver/media_driver/linux/gen11/codec/ddi -I/media-driver/media_driver/linux/gen11/ddi -I/media-driver/../gmmlib/Source/inc -I/media-driver/../gmmlib/Source/inc/common -I/media-driver/../gmmlib/Source/inc/platform/iAlm -I/media-driver/../gmmlib/Source/inc/umKmInc -I/media-driver/../gmmlib/Source/GmmLib/inc -I/media-driver/../huc/inc -I/linux -Wno-reorder -Wsign-promo -Wnon-virtual-dtor -Wno-invalid-offsetof -fvisibility-inlines-hidden -fno-use-cxa-atexit -frtti -fexceptions -fpermissive -fcheck-new -std=c++1y -std=c++11 -O2 -DNDEBUG -fPIC -Werror -Wall -Winit-self -Wpointer-arith -Wno-unused -Wno-unknown-pragmas -Wno-endif-labels -Wno-comments -Wno-sign-compare -Wno-attributes -Wno-narrowing -Wno-overflow -Wno-parentheses -Werror=implicit-function-declaration -Werror=address -Werror=format-security -Werror=non-virtual-dtor -Werror=return-type -march=corei7 -mpopcnt -msse -msse2 -msse3 -mssse3 -msse4.1 -msse4.2 -msse4 -mfpmath=sse -finline-functions -fno-short-enums -Wa,--noexecstack -fno-strict-aliasing -DUSE_MMX -DUSE_SSE -DUSE_SSE2 -DUSE_SSE3 -DUSE_SSSE3 -fmessage-length=0 -fstack-protector -fdata-sections -ffunction-sections -m64 -DLINUX=1 -DLINUX -DNO_RTTI -DNO_EXCEPTION_HANDLING -DINTEL_NOT_PUBLIC -g -fvisibility=hidden -funswitch-loops -Wl,--gc-sections -D_AMD64_ -D__CT__ -D__linux__ -fPIC -fno-tree-pre -Wl,--no-as-needed -O2 -D_FORTIFY_SOURCE=2 -fno-omit-frame-pointer -finline-limit=100 -o CMakeFiles/iHD_drv_video_OBJ.dir/agnostic/common/cm/cm_device_rt.cpp.o -c /media-driver/media_driver/agnostic/common/cm/cm_device_rt.cpp`

Pay attention that there is:
1. This comes from gmmlib pkg-config:
`-I/include/igdgmm/util -I/include/igdgmm/inc/common -I/include/igdgmm/inc -I/include/igdgmm/GmmLib/inc -I/include/igdgmm/GmmLib -I/include/igdgmm`
2. And this comes somewhere from the driver cmake files:
`-I/media-driver/../gmmlib/Source/inc -I/media-driver/../gmmlib/Source/inc/common -I/media-driver/../gmmlib/Source/inc/platform/iAlm -I/media-driver/../gmmlib/Source/inc/umKmInc -I/media-driver/../gmmlib/Source/GmmLib/inc`

If gmmlib was found via pkg-config the 2nd list of paths should _not_ present.

@chivakker: FYI

chivakker, 2018-09-13:
Also, please re-arrange all the including folders and groups everything under one single place.  Define a clear public API and adhere to it.

It is a major overhauling of the whole building system but doing it now will help to ease major problems in the future.

ljia5, 2020-06-04:
Fix patch merged, please close this issue.

closed 2020-07-03 by XinfengZhang