intel/media-driver #317: [KBL][ffmpeg-vaapi][vp8] decoding fails

fulinjie, 2018-09-14:
cmd line: 
ffmpeg -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -v verbose -y -i ./vp80-02-inter-1418.ivf -f md5 ffmpeg_vaapi.md5

media driver:
commit id:
6ecb1d70f501576fe9dd80f580e40e59a22dab61

src file:
[https://github.com/webmproject/vp8-test-vectors/blob/master/vp80-02-inter-1418.ivf](url)

error message:
[vp8 @ 0x55a85df70d40] Failed to end picture decode issue: 23 (internal decoding error).
Error while decoding stream #0:0: Input/output error
[vp8 @ 0x55a85df73e40] Failed to end picture decode issue: 23 (internal decoding error).
[vp8 @ 0x55a85df76f40] Failed to end picture decode issue: 23 (internal decoding error).
Error while decoding stream #0:0: Input/output error
    Last message repeated 1 times
[vp8 @ 0x55a85df603c0] Failed to end picture decode issue: 23 (internal decoding error).
Error while decoding stream #0:0: Input/output error
[vp8 @ 0x55a85df617c0] Failed to end picture decode issue: 23 (internal decoding error).
Error while decoding stream #0:0: Input/output error
[vp8 @ 0x55a85df64940] Failed to end picture decode issue: 23 (internal decoding error).
Error while decoding stream #0:0: Input/output error
    Last message repeated 6 times
No more output streams to write to, finishing.
frame=   79 fps=0.0 q=-0.0 Lsize=       0kB time=00:00:03.60 bitrate=   0.1kbits/s speed=  49x    
video:4629kB audio:0kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: unknown
Input file #0 (./vp80-02-inter-1418.ivf):
  Input stream #0:0 (video): 108 packets read (109579 bytes); 79 frames decoded; 
  Total: 108 packets (109579 bytes) demuxed
Output file #0 (ffmpeg_vaapi.md5):
  Output stream #0:0 (video): 79 frames encoded; 79 packets muxed (4740000 bytes); 
  Total: 79 packets (4740000 bytes) muxed

This file could be decoded successfully on i965 driver.

wangyan42164, 2018-09-18:
I tried this patch in the latest ffmpeg on KBL. The error is gone with the source file.

uartie, 2018-09-18:
works for me!

closed 2019-08-29 by wangyan-intel

uartie, 2020-09-21:
FFMpeg patch never got merged to upstream... see #1028