intel/media-driver #320: [gstreamer-vaapi][hevc-10bit-enc]encoding fail with iHD driver

wangzj0601, 2018-09-18:
env:
platform: kbl
libva: 30c751f7b74c43021f33361c1baf2aa92acd0317
media-driver: 888798f836f418cc4171c61e0050f09955a2ce6a
gst-vaapi: 7b077cdcb96864726ccf682bed03384aaa4f7588

Step
run below cmd 
gst-launch-1.0 videotestsrc num-buffers=10 ! video/x-raw,format=P010_10LE ! vaapih265enc ! filesink location=aa.h265

error message:
Setting pipeline to PAUSED ...
Pipeline is PREROLLING ...
Got context from element 'vaapiencodeh265-0': gst.vaapi.Display=context, gst.vaapi.Display=(GstVaapiDisplay)"\(GstVaapiDisplayX11\)\ vaapidisplayx11-1";
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
0:00:00.041160687  7607      0x1bf7990 ERROR                  vaapi gstvaapiencoder.c:541:gst_vaapi_encoder_put_frame: failed to encode frame (status = -1)
0:00:00.041265449  7607      0x1bf7990 ERROR            vaapiencode gstvaapiencode.c:722:gst_vaapiencode_handle_frame: failed to encode frame 1 (status -1)
ERROR: from element /GstPipeline:pipeline0/GstVideoTestSrc:videotestsrc0: Internal data stream error.
Additional debug info:
gstbasesrc.c(3055): gst_base_src_loop (): /GstPipeline:pipeline0/GstVideoTestSrc:videotestsrc0:
streaming stopped, reason error (-5)

This cmd can works well with i965 driver.

wangyan42164, 2018-09-28:
@XinfengZhang 
This bug should come from CodechalEncHevcStateG9::Encode8x8PBMbEncKernel() when it calls CodechalEncHevcStateG9::AddCurbeToStateHeap().

The root cause is that dashSize is based on kernelState->KernelParams.iCurbeLength which isn't initialized in  CodechalEncHevcStateG9::InitKernelStateMbEnc().

For P010_10LE, 
m_numMbEncEncKrnStates = CODECHAL_HEVC_MBENC_NUM_KBL;

For NV12, 
m_numMbEncEncKrnStates = CODECHAL_HEVC_MBENC_NUM_BXT_SKL;

So for P010_10LE, CODECHAL_HEVC_MBENC_PENC kernel state isn't initialized.
When CodechalEncHevcStateG9::InitKernelStateMbEnc() selects CODECHAL_HEVC_MBENC_PENC for kernelState, the block size isn't enough for curbe data.

We need clarify how to select kernelState for 10bit P frame encoding in KBL platform.
After this, I can submit a PR for fixing it.
Thanks.
Yan Wang

wangyan42164, 2018-10-09:
@XinfengZhang Could you please review my PR for fixing this bug? Thanks.

wangyan42164, 2018-10-19:
Previous PR https://github.com/intel/media-driver/pull/333 will cause ULT error when manual build on WIndows X64 Debug. So I refined it as #371.

PR #371 (2018-12-06): Fix HEVC 10bit encoding on KBL. (GitHub issue #320)
1. Make sure CODECHAL_HEVC_MBENC_PENC kernel state initialized.
2. Add support format for 10bit.
3. Make sure CODECHAL_HEVC_MBENC_DS_COMBINED kernel state initialized.
But need disable it on BXT and SKL. Check kernel size in
CodechalEncHevcStateG9::InitKernelStateMbEnc() instead of hard code.

Fixes #320.
closed 2018-12-06 by Sherry-Lin