intel/media-driver #322: [gstreamer-vaapi][ICL] vaQuerySurfaceAttributes doesn't support YUY2/UYVY VASurfaceAttribPixelFormat.

feiwan1, 2018-09-19:
Now, vaQuerySurfaceAttributes only return NV12 format with attrib type of VASurfaceAttribPixelFormat.
Need your help to also support YUY2/UYVY to meet Iotg requirement.

XinfengZhang, 2018-09-25:
which context? VP, ENCODE,DECODE?

feiwan1, 2018-09-25:
Hi @XinfengZhang ， Encode.

wangyan42164, 2018-10-09:
@feiwan1 @XinfengZhang I submitted one PR for this previously. Could you please review it? Thanks.

feiwan1, 2018-10-10:
Hi @wangyan42164 , Did you mean PR #323? It should be the fix of issue #318, not the fix of this issue.

wangyan42164, 2018-10-11:
@feiwan1 How should I reproduce this issue? What profile do you use? VAProfileHEVCMain?

feiwan1, 2018-10-12:
Hi @wangyan42164 , I used gst-vaapi cmdline to reproduce this issue:
# gst-launch-1.0 videotestsrc num-buffers=100 ! video/x-raw,format=YUY2 ! vaapih265enc ! fakesink

The default profile is VAProfileHEVCMain, does this profile support YUY2 in driver side? Or I need to use  
other profile like 422-main-10?

wangyan42164, 2018-10-12:
@feiwan1 I did some initial checking on my KBL(Gen9) platform because I have not ICL (Gen11). But I think they are similar in this issue.
It should have 2 points for this issue:
1. Surface attribute of VAProfileHEVCMain in media driver doesn't include YUY2 and UYVY. I added some code to fix it.
2. gstreamer-vaapi cannot support it. I use master branch of gstreamer-vaapi build like the following:
gst-launch-1.0 version 1.15.0
GStreamer 1.15.0 (GIT)
git commit: 00f5fe9ccfc7183eb0f3f02f6efcca06343e70b2

It seems cannot support YUY2 based on the following log:
0:00:00.021939639 16404      0x191d990 ERROR                  vaapi gstvaapiencoder.c:812:is_chroma_type_supported: We only support YUV 4:2:0 and YUV 4:2:2 for encoding. Please try to use vaapipostproc to convert the input format.

If necessary, I can submit PR for adding YUY2 and UYVY surface support in media driver. 
Thanks.

feiwan1, 2018-10-12:
@wangyan42164 , please add YUY2/UYVY surface support in meida-driver. I will take a look at gst-vaapi side.
Thanks

wangyan42164, 2018-10-12:
@feiwan1 Added. Thanks.

FurongZhang, 2018-10-17:
I read the whole thread, it is related to codec context. @XinfengZhang to check it.

feiwan1, 2018-10-18:
According @Maggieie comment on #359 , VAProfileHEVCMain will not support YUY2/UYVY format, If so, do we need to support VAProfileHEVCMain422_10 encode? Now vainfo shows only VAProfileHEVCMain422_10 decode is supported on ICL.

wangyan-intel, 2019-08-29:
Based on @Maggieie's comments, we don't support it.

closed 2019-08-29 by wangyan-intel