intel/media-driver #325: [ffmpeg-qsv][avc_encode] avc cbr/vbr encode file bitrate out of range target 10% on iHD driver.

zhuqingliang, 2018-09-20:
environment:
SKL OS: ubuntu 16.04 kernel: 4.15.0 
libva b6c50dad4d7d14c507108d9f468662e2d7ae1c4d ​https://github.com/01org/libva.git 
iHD Driver ed04556a6676fd77ff845508bc54bfc60915450c ​https://github.com/intel/media-driver
libva-utils 9a10ad663349732decd668a426cfc349b0010d1d ​https://github.com/01org/libva-utils.git
Mediasdk 7ea683585f0f84f1bea035a405855312c92c46c0 ​https://github.com/Intel-Media-SDK/MediaSDK
FFMPEG 56f68a099cc607658118e00cad30569103ae3751 ​https://git.ffmpeg.org/ffmpeg.git

How to reproduce:
1.build env as above lists
2.ffmpeg -hwaccel qsv -init_hw_device qsv=hw -filter_hw_device hw -v verbose -f rawvideo -video_size 352x288 -pix_fmt yuv420p -i ./MOBILE_352x288_30_orig_01.yuv -an -vf hwupload=extra_hw_frames=64,format=qsv -c:v h264_qsv -g 30 -bf 3 -r 30 -b:v 50k -maxrate:v 50k -preset 4 -vframes 100 -y ./h264e_cbr_352x288_gop30_refdist3_bitrate100k.h264
3. size of h264e_cbr_352x288_gop30_refdist3_bitrate100k.h264 =25550 
4. bitrate_real=25550*8*30/1024/100 =  59.88
5. bitrate_input=50
6.. bitrate_gap = (59.88-50)/50= 19.00% large than 10%

failed case:
h264e_cbr_352x288_gop30_refdist3_bitrate100k.h264 on CFL/KBL/SKL
h264e_cbr_3840x2160_gop30_refdist3_bitrate3M.h264 on CFL/KBL/SKL
h264e_cbr_3840x2160_gop30_refdist3_bitrate5M.h264 on SKL
h264e_cbr_720x480_gop30_refdist3_bitrate1M.h264 on CFL/KBL
h264e_cbr_720x480_gop30_refdist3_bitrate2M.h264 on CFL/KBL
h264e_vbr_352x288_gop30_refdist3_bitrate100k.h264 on CFL/KBL/SKL
h264e_vbr_352x288_gop30_refdist3_bitrate200k.h264 on CFL/KBL/SKL
h264e_vbr_720x480_gop30_refdist3_bitrate1M.h264 on CFL/KBL/SKL
h264e_vbr_1280x720_gop30_refdist3_bitrate8M.h264 on CFL/KBL/SK

wangyan42164, 2018-10-15:
@zhuqingliang @XinfengZhang I can reproduce it in my KBL GT2 by MOBILE_352x288_30_orig_01.yuv.
But this issue may come from MediaSDK and not media driver.
The following is the detail:
MfxHwH264Encode::CheckVideoParamQueryLike() in MediaSDK/_studio/mfx_lib/shared/src/mfx_h264_enc_common_hw.cpp:3397:
                mfxF64 rawDataBitrate = 12.0 * par.mfx.FrameInfo.Width * par.mfx.FrameInfo.Height *
                par.mfx.FrameInfo.FrameRateExtN / par.mfx.FrameInfo.FrameRateExtD;
                mfxU32 minTargetKbps = mfxU32(MFX_MIN(0xffffffff, rawDataBitrate / 1000 / 700));

                if (par.calcParam.targetKbps < minTargetKbps)
                {
                    changed = true;
                    par.calcParam.targetKbps = minTargetKbps;
                }

par.mfx.FrameInfo.Width=352
par.mfx.FrameInfo.Height=288
par.mfx.FrameInfo.FrameRateExtN=30
par.mfx.FrameInfo.FrameRateExtD=1

So minTargetKbps=52 > 50, the final targetKbps is set into 52 and not 50.
And I get encoded stream is 27728 bytes and bitrate_real=27728 * 8 * 30 / 1024 / 100 = 64.9875 bitrate_gap = (64.9875-50)/50= 29.975% large than 10%

So when I use 800 instead of 700, the final targetKbps is set into 50.
And I get encoded stream is 22886 bytes and bitrate_real=22886 * 8 * 30 / 1024 / 100 = 53.639 bitrate_gap = (53.639-50)/50= 7.28% is OK.

So should I submit one PR for MediaSDK?
Thanks.

wangyan42164, 2018-10-15:
After talked with @XinfengZhang, I will submit a PR to MediaSDK git upstream. Thanks.

lizhong1008, 2018-11-21:
@wangyan42164 ， any update? Woud you please provide the MSDK issue link?

wangyan42164, 2018-11-21:
@lizhong1008 No more response from MSDK team. You can refer to the discussion in the above #804.

onabiull, 2020-05-28:
@zhuqingliang are you ok to close this issue? If it is still relevant looks https://github.com/Intel-Media-SDK/MediaSDK are better place for it.

closed 2020-06-11 by XinfengZhang