intel/media-driver #328: [ICLLP][AVC lowe power Encode] AVC low power encode brc fail due to huc loading fail

emmaqiaogn, 2018-09-26:
build and install https://github.com/intel/media-driver
build https://github.com/intel/libva-utils

mv_encoder_adv -i cityLetters_720x480_30_1_500.yuv -o cityLetters_720x480_30_1_500_mvencadv.264 --file-fmt i420 --encode avc --SourceWidth 720 --SourceHeight 480 --NumFrames 500 --RefFrameType 2 --FrameRate 29.97 --EncoderInput NV12 --EncoderOutput NV12 --BitRate 11099889 --MaxBitRate 0 --MinBitRate 11099889 --TargetBitRate 11099889 --RateControl 1 --TargetUsage 1 --MAD 0 --MADInterval 1 --AVBRAccuracy 100 --AVBRConvergence 1 --InstantBitRateCheck 0 --InstantBitRateAccuracy 250 --InstantBitRateConvergence 0 --GOPBRCResetInterval 0 --vdenc --PictureCodingType 0 --EntropyCodingMode 0 --CabacInitIDC 1 --DeblockingIDC 0 --ProfileIDC 77 --LevelIDC 32 --ChromaFormatIDC 1 --GOPPicSize 36 --GOPRefDist 1 --BPyramid 0 --NumRefFrames 2 --SliceStructure 2 --NumSlicesResetInterval 0 --NumSlices 30 --NumSlicesForReset 1,1,1,1,1,1,1,1 --PSliceType 5 --BSliceType 6 --ISliceType 7 --ISliceQP 25 --PSliceQP 25 --BSliceQP 25 --SliceQpDelta 0 --MBBRC 1 

ERROR   :FrameModeWrapper: [trans] ERROR:ENCODE: attribute not supported(issue #1)
ERROR   :FrameModeWrapper: [trans] ERROR:ENCODE: Encoder failed during configuration
ERROR   :FrameModeWrapper: [trans] ERROR:COMMON: Fail to open encode accelerator
ERROR   :FrameModeWrapper: [trans] ERROR:COMMON: Builder is failed... deleting created media components
ERROR   :FrameModeWrapper: [trans] ERROR:COMMON: Fail to create Encoder Product

emmaqiaogn, 2018-09-26:
set boot/grub.cfg
i915.enable_guc=3

EwoutH, 2019-05-20:
Any update on this issue?

XinfengZhang, 2020-05-14:
@emmaqiaogn , could we close the issue

XinfengZhang, 2020-06-07:
close it, it could not reproduced on latest kernel version , such as 5.4

closed 2020-06-07 by XinfengZhang