intel/media-driver #330: Compilation error when compiled with Tag: 18.2.0

nehudesi, 2018-09-26:
I am getting below error when compiled with 18.2.0 tag:

clang-7: error: unknown argument: '-fno-tree-pre'
clang-7: error: unknown argument: '-fno-tree-pre'
clang-7: warning: optimization flag '-funswitch-loops' is not supported [-Wignored-optimization-argument]
make[2]: *** [gmmlib/CMakeFiles/gmm_umd.dir/build.make:63: gmmlib/CMakeFiles/gmm_umd.dir/build/whlrvp/tmp/portage/x11-libs/intel-media-18.2.0/work/gmmlib-5ff84b923ec2ba1572c464dc8def73348571b440/Source/Common/AssertTracer/AssertTracer.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
clang-7: error: unknown argument: '-fno-tree-pre'
clang-7: warning: optimization flag '-funswitch-loops' is not supported [-Wignored-optimization-argument]
make[2]: *** [gmmlib/CMakeFiles/gmm_umd.dir/build.make:183: gmmlib/CMakeFiles/gmm_umd.dir/CachePolicy/GmmGen10CachePolicy.cpp.o] Error 1

**With same environment, build succeeds when checkout commit with commit 7f2fbc154e05cf8357acffe2dfe1c296b59406ae**

Not sure, what had changed between 7f2fbc154e05cf8357acffe2dfe1c296b59406ae and tag 18.2.0.

nehudesi, 2018-10-04:
With 18.3, I am getting below error:

```
../../media-driver-intel-media-18.3.0/media_driver/linux/common/ddi/media_libva.h:213:5: error: unknown type name 'VASliceParameterBufferHEVCExtension'; did you mean 'VASliceParameterBufferHEVC'?
    VASliceParameterBufferHEVCExtension          *pVASliceParaBufHEVCRext;
    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    VASliceParameterBufferHEVC
../../../../../../../usr/include/va/va_dec_hevc.h:328:3: note: 'VASliceParameterBufferHEVC' declared here
} VASliceParameterBufferHEVC;
  ^
In file included from ../../media-driver-intel-media-18.3.0/media_driver/linux/common/codec/ddi/media_ddi_decode_base.cpp:27:
In file included from ../../media-driver-intel-media-18.3.0/media_driver/linux/common/codec/ddi/media_libva_decoder.h:30:
../../media-driver-intel-media-18.3.0/media_driver/linux/common/ddi/media_libva.h:219:5: error: unknown type name 'VAPictureParameterBufferHEVCExtension'; did you mean 'VAPictureParameterBufferHEVC'?
    VAPictureParameterBufferHEVCExtension        PicParamHEVCRext;
    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    VAPictureParameterBufferHEVC
../../../../../../../usr/include/va/va_dec_hevc.h:195:3: note: 'VAPictureParameterBufferHEVC' declared here
} VAPictureParameterBufferHEVC;
  ^
../../media-driver-intel-media-18.3.0/media_driver/linux/common/codec/ddi/media_ddi_decode_base.cpp:744:14: error: use of undeclared identifier 'VASubsetsParameterBufferType'; did you mean 'VASliceParameterBufferType'?
        case VASubsetsParameterBufferType:
             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
             VASliceParameterBufferType
../../../../../../../usr/include/va/va.h:1533:5: note: 'VASliceParameterBufferType' declared here
    VASliceParameterBufferType          = 4,
    ^
../../media-driver-intel-media-18.3.0/media_driver/linux/common/codec/ddi/media_ddi_decode_base.cpp:744:14: error: duplicate case value 'VASliceParameterBufferType'
        case VASubsetsParameterBufferType:
             ^
../../media-driver-intel-media-18.3.0/media_driver/linux/common/codec/ddi/media_ddi_decode_base.cpp:732:14: note: previous case defined here
        case VASliceParameterBufferType:
             ^
4 errors generated.

```

Sherry-Lin, 2018-10-09:
Have you update the libva version to 2.3.0? You can refer to Tools/dockers/ubuntu/18.04/Dockerfile to check your environment.

nehudesi, 2018-10-09:
Thank you for your reply! I will try to do that and will update here.

nehudesi, 2018-10-11:
Ok.. With libva 2.3.0 I am able to pass through this error. However, I now hitting this error:

/codec/ddi/media_libva_decoder.cpp:45:10: fatal error: 'X11/Xutil.h' file not found
#include <X11/Xutil.h>
         ^~~~~~~~~~~~~
1 error generated.
make[2]: *** [media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/build.make:9711: media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/linux/common/codec/ddi/media_libva_decoder.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....

XinfengZhang, 2019-08-29:
already add protection code for #include <X11/Xutil.h>
now the code is
#if !defined(ANDROID) && defined(X11_FOUND)
#include <X11/Xutil.h>
#endif

closed 2019-08-29 by XinfengZhang