intel/media-driver #334: [regression] ffmpeg-vaapi wrong chroma output in jpeg decode 444P image

uartie, 2018-10-01:
First bad commit:

```
commit 53d304115c3faaa3a2c91c16af079214ebc7d9e1 (HEAD -> master)
Author: xinfeng_(carl)__zhang <carl.zhang@intel.com>
Date:   Sat Sep 29 11:54:52 2018 +0800

    [Decode] call vp in the vaPutImage(CL#803612)
    
    1. set image to be a linear surface
    2. use VP to copy the surface into image
    3. avoid read slowly of tile surface(map_gtt)
    
    Change-Id: I9badccb436d3c6dc17ac90bb89a3936f39a017d4
```

LIBVA_DRIVER_NAME=iHD ffmpeg -hwaccel vaapi -hwaccel_device /dev/dri/renderD128 -v verbose -i smpte_444P.jpeg -pix_fmt yuv444p -f rawvideo -vsync passthrough -vframes 1 -y smpte_444P_720x480.yuv

uartie, 2018-10-01:
**Expected Result:**
[image]

**Actual Result:**
_Y'CbCr_
[image]
_Cb Only_
[image]
_Cr Only_
[image]

uartie, 2018-10-01:
JPEG 4:4:4 Input File:
[image]

XinfengZhang, 2018-10-06:
yes, we know the root cause, will revert the change firstly, then check in again after fix these issues

XinfengZhang, 2018-10-06:
root cause is, get image will call VP to copy data from tile surface to linear surface, but it call some other function to scale the output surfaces

closed 2019-08-29 by XinfengZhang