intel/media-driver #338: WTF happen？

adaaaaaa, 2018-10-03:
error code here：
[ 77%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/linux/common/cm/media_libva_cm.c.o
[ 78%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/linux/common/codec/ddi/media_ddi_decode_base.cpp.o
/tmp/pamac-build-captain/intel-media-driver/src/media-driver-intel-media-18.3.0/media_driver/linux/common/codec/ddi/media_ddi_decode_base.cpp: 在成员函数‘virtual VAStatus DdiMediaDecode::CreateBuffer(VABufferType, uint32_t, uint32_t, void*, VABufferID*)’中:
/tmp/pamac-build-captain/intel-media-driver/src/media-driver-intel-media-18.3.0/media_driver/linux/common/codec/ddi/media_ddi_decode_base.cpp:744:14: 错误：‘VASubsetsParameterBufferType’在此作用域中尚未声明
         case VASubsetsParameterBufferType:
              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/tmp/pamac-build-captain/intel-media-driver/src/media-driver-intel-media-18.3.0/media_driver/linux/common/codec/ddi/media_ddi_decode_base.cpp:744:14: 附注：suggested alternative: ‘VASliceParameterBufferType’
         case VASubsetsParameterBufferType:
              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
              VASliceParameterBufferType
make[2]: *** [media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/build.make:5263：media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/linux/common/codec/ddi/media_ddi_decode_base.cpp.o] 错误 1
make[1]: *** [CMakeFiles/Makefile2:170：media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/all] 错误 2
make: *** [Makefile:152：all] 错误 2
==> 错误： 在 build() 中发生一个错误。
    正在放弃...

XinfengZhang, 2018-10-06:
please update the libva, we add subset parameter recently.

uartie, 2018-10-19:
@XinfengZhang should configuration files bump minimum required libva version to prevent getting this far into compilation?

closed 2019-08-29 by XinfengZhang