intel/media-driver #343: usleep/polling synchronization in media driver

dmitryermilov, 2018-10-05:
Manually inspecting code of the driver, I found that inside usleep() calls used  for synchronization inside vaMapBuffer() :
https://github.com/intel/media-driver/blob/efcce467c29fe3d793a052acbe062de115ded2d6/media_driver/linux/common/codec/ddi/media_ddi_encode_base.cpp#L250 

This looks suspicious.
- this synchronization part is implemented via polling:
```
            // Wait until encode PAK complete, sometimes we application detect encoded buffer object is Idle, may Enc done, but Pak not.
            uint32_t maxTimeOut                               = 100000;  //set max sleep times to 100000 = 1s, other wise return error.
            if (timeOutCount < maxTimeOut)
            {
                //sleep 10 us to wait encode complete, it won't impact the performance.
                uint32_t sleepTime                            = 10;      //sleep 10 us when encode is not complete.
                usleep(sleepTime);
                timeOutCount++;
                continue;
            }
```
Polling is usually a bad practice because it significantly increases CPU usage.
- usleep gives no warranties about max time of execution. It only guaranties that usleep will sleep not less than the argument specified.  

In general, it seems to be a workaround. Can you please clarify the purpose of usleep in DdiEncodeBase::StatusReport() and consider a proper implementation? Is it possible to implement it via some ioctl(WAIT) call ?

dmitryermilov, 2018-10-05:
+ @dvrogozh

dmitryermilov, 2018-10-10:
@XinfengZhang , can you please comment or assign this to someone?

LuoIvan, 2018-10-11:
@dmitryermilov , thanks for comments. GetstatusReport provides one way  for app to proactive check the readiness of specific task at GPU side. So as long as no other event notification mechanism enabled on Linux, non-pooling like usplee is needed.
here is  the POSIX specification of usleep 
The usleep() function will cause the calling thread to be suspended from execution ...
so this clearly requires that it suspends execution and lets the resources to other processes or threads.

eero-t, 2018-10-18:
10us sleep is *very* short.  Polling with microscopic sleeps definitely uses CPU because scheduling itself has some overhead: switching to kernel context, calculating which process to run next and switching to that.

(I think Linux normally schedules process at most at 1000Hz, i.e. if other processes are waiting to be woken, each such process can cause 1ms delay until your process is woken again, depending on the priorities of the involved processes.)

Xiaogangli-intel, 2020-03-24:
Actually, AVC encode has another way to update status report, in this way, usleep is not needed. We are working for HEVC currently.
For other codecs, usleep is needed, especially for multi-PAKs cases. Actually we do called ioctl(WAIT) before this usleep, but because implementation of status report in codechal, after ioctl(WAIT) finished, we cannot ensure that staus report had been updated, we need a short time to wait status report be updated.

Xiaogangli-intel, 2020-03-26:
Not sure if still have other concerns, if no, I will close this issue.

dmitryermilov, 2020-03-26:
> we cannot ensure that staus report had been updated, we need a short time to wait status report be updated

This is exactly what this issue is about. Now you confirm the issue exists. 
Why are you willing to close it without a corresponding fix?

Xiaogangli-intel, 2020-03-26:
Hello Dima, this is not an issue, this just the implementation in the driver, there are limitations. We will improve it step by step, but it should not be an issue. We also want to use the event to handle this, then we don't need polling, but it's not supported currently, right?

dmitryermilov, 2020-03-26:
Hi Xiaogang,

Let's look at this from another angle: do you agree that **generally** (sometimes polling is better than expansive ioctl call but it looks like it's not the case here) polling is bad  ? And currently the polling exists. So from my perspective it's an issue. On the other hand, limitations/known issues occurs. There is nothing really terrible about it. Presence of current issue being in an "open" state is just a description of current situation.

Xiaogangli-intel, 2020-03-26:
Yes, it's ok for me.

Xiaogangli-intel, 2021-09-22:
close it because we will not fix it.

closed 2021-09-22 by Xiaogangli-intel