intel/media-driver #356: media-driver 18.0.3 - compilation fails on 32bit

kwizart, 2018-10-11:
I'm trying to compile on media-driver 18.0.3 on i686 and I have a compilation error.
For the record, I still consider important to have a 32bit for running steam client on x86_64 OS.

Full compilation log: http://koji.rpmfusion.org/kojifiles/work/tasks/6969/266969/build.log

---
cc1 -m32 -march=i686 -mtune=generic -msse2 -mfpmath=sse -mstackrealign -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -Wno-reorder -Wsign-promo -Wnon-virtual-dtor -Wno-invalid-offsetof -fvisibility-inlines-hidden -fno-use-cxa-atexit -frtti -fexceptions -fpermissive -fcheck-new -std=c++1y -std=c++11 -DNDEBUG -fPIC   -Werror -Wall -Winit-self -Wpointer-arith -Wno-unused -Wno-unknown-pragmas -Wno-comments -Wno-sign-compare -Wno-attributes -Wno-narrowing -Wno-overflow -Wno-parentheses -Werror=implicit-function-declaration -Werror=address -Werror=format-security -Werror=non-virtual-dtor -Werror=return-type -march=corei7 -mpopcnt -msse -msse2 -msse3 -mssse3 -msse4.1 -msse4.2 -msse4 -mfpmath=sse -finline-functions -fno-short-enums -Wa,--noexecstack -fno-strict-aliasing -DUSE_MMX -DUSE_SSE -DUSE_SSE2 -DUSE_SSE3 -DUSE_SSSE3 -fmessage-length=0 -fstack-protector -fdata-sections -ffunction-sections -m32 -DLINUX=1 -DLINUX -DNO_RTTI -DNO_EXCEPTION_HANDLING -DINTEL_NOT_PUBLIC -g -fvisibility=hidden -funswitch-loops -Wl,--gc-sections -D__linux__ -fPIC -fno-tree-pre -Wl,--no-as-needed -O2 -D_FORTIFY_SOURCE=2 -fno-omit-frame-pointer -finline-limit=100 -o CMakeFiles/iHD_drv_video_OBJ.dir/agnostic/common/cm/cm_buffer_rt.cpp.o -c /builddir/build/BUILD/media-driver-intel-media-18.3.0/media_driver/agnostic/common/cm/cm_buffer_rt.cpp
make[2]: Leaving directory '/builddir/build/BUILD/media-driver-intel-media-18.3.0/build'
In file included from /builddir/build/BUILD/media-driver-intel-media-18.3.0/media_driver/linux/common/os/libdrm/include/xf86drm.h:41,
                 from /builddir/build/BUILD/media-driver-intel-media-18.3.0/media_driver/linux/common/ddi/media_libva_common.h:33,
                 from /builddir/build/BUILD/media-driver-intel-media-18.3.0/media_driver/linux/common/cm/media_libva_cm.h:30,
                 from /builddir/build/BUILD/media-driver-intel-media-18.3.0/media_driver/linux/common/cm/cm_def_os.h:41,
                 from /builddir/build/BUILD/media-driver-intel-media-18.3.0/media_driver/agnostic/common/cm/cm_def.h:30,
                 from /builddir/build/BUILD/media-driver-intel-media-18.3.0/media_driver/agnostic/common/cm/cm_buffer.h:30,
                 from /builddir/build/BUILD/media-driver-intel-media-18.3.0/media_driver/agnostic/common/cm/cm_buffer_rt.h:30,
                 from /builddir/build/BUILD/media-driver-intel-media-18.3.0/media_driver/agnostic/common/cm/cm_buffer_rt.cpp:27:
/builddir/build/BUILD/media-driver-intel-media-18.3.0/media_driver/linux/common/os/libdrm/include/libdrm_macros.h: In function 'int drm_munmap(void*, size_t)':
/builddir/build/BUILD/media-driver-intel-media-18.3.0/media_driver/linux/common/os/libdrm/include/libdrm_macros.h:40:40: error: size of array is negative
       (void) sizeof(char [1 - 2*!(COND)]); \
                                        ^
/builddir/build/BUILD/media-driver-intel-media-18.3.0/media_driver/linux/common/os/libdrm/include/libdrm_macros.h:76:4: note: in expansion of macro 'STATIC_ASSERT'
    STATIC_ASSERT(LARGE_OFF_T % 2147483629 == 721 &&
    ^~~~~~~~~~~~~
make[2]: *** [media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/build.make:79: media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/agnostic/common/cm/cm_buffer_rt.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:245: media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/all] Error 2
make: *** [Makefile:155: all] Error 2
---

dvrogozh, 2019-02-19:
Interesting that Debian did not meet with this issue and managed to package media-driver in both 32- and 64-bit flavors. They applied a custom patch however: https://salsa.debian.org/multimedia-team/intel-media-driver/blob/master/debian/patches/0002-Remove-settings-based-on-ARCH.patch, but I don't think that it relates to the issue you describe.

I had no chance to try build myself yet for Fedora/CentOS, but thought it worths to provide the above info right away.

dvrogozh, 2019-02-21:
@[kwizart](/kwizart): you could wish to join discussion here: https://github.com/intel/libva/issues/281

jbeich, 2019-04-10:
Also doesn't affect FreeBSD i386 where `off_t` is always 64-bit.
http://beefy5.nyi.freebsd.org/data/latest-per-pkg/libva-intel-media-driver/

XinfengZhang, 2020-03-09:
@kwizart how about the status,  (void) sizeof(char [1 - 2*!(COND)]) should be possible negative both on 32 and 64 bit, the build check option is different?

kwizart, 2020-03-09:
@XinfengZhang I don't know why it fails only on 32bit and not on 64bit.

dvrogozh, 2020-03-09:
@kwizart : the code (starting from) https://github.com/intel/media-driver/blob/60c0a69a83c742dccadd7c8bc93d406542f73b18/media_driver/linux/common/os/i915/include/libdrm_macros.h#L68 which fails to compile on your side is still a copy from https://gitlab.freedesktop.org/mesa/drm/-/blob/master/libdrm_macros.h#L70. So, I wonder, does libdrm still can be successfully compiled on 32-bit Fedora? And if yes, why? It seems the static assert check tries to verify that off_t is 64-bit and it fails... What's sizeof(off_t) on Fedora 32-bit? Are there any compilation flags required to fetch correct off_t defition?

kwizart, 2020-03-09:
It's interesting that there is mention of LARGE FILE support and on linux x86 32bit there is indeed a need to enable a special CXXFLAGS such as -D_FILE_OFFSET_BITS=64 or by default the value is 32.

I  confirm that now that this define is set I have a different error (probably unrelated as it might also have failed on x86_64) (1)

I've checked with fedora libdrm, and this flag isn't set explicitly in the package, so it might be autodetected as appropriate. But for the libva case, I don't see this flag been set, so I might have to enforce it. (and even maybe to all downstream users of the vaapi strack ?).

(1)
> make[2]: Leaving directory '/builddir/build/BUILD/media-driver-intel-media-19.4.0/build'
> /builddir/build/BUILD/media-driver-intel-media-19.4.0/media_driver/linux/common/os/i915/xf86drm.c: In function 'int drmOpenByBusid(const char*, int)':
> /builddir/build/BUILD/media-driver-intel-media-19.4.0/media_driver/linux/common/os/i915/xf86drm.c:606:15: error: '%s' directive argument is null [-Werror=format-overflow=]
>   606 |         drmMsg("drmOpenByBusid: drmGetBusid reports %s\n", buf);
>       |         ~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
> cc1plus: all warnings being treated as errors

dvrogozh, 2020-03-09:
> I've checked with fedora libdrm, and this flag isn't set explicitly in the package

I also don't see it's set explicitly. Though I always though that's the flag which needs explicit setting. Maybe there is some special meson setting which triggers it's being set implicitly and we miss it in a grep. Anyway, I think it worth to be added permanently to the driver build flags.

Yeah, the second issue looks unrelated.

kwizart, 2022-10-24:
Compilation has succeeded with current intel-media-driver releases with -D_FILE_OFFSET_BITS=64 explicitly set.

I haven't verified if it's set by default nowadays....

kwizart, 2022-11-29:
CLosing the issue

closed 2022-11-29 by kwizart