intel/media-driver #361: [ICL][HEVC Encode][Low power]vaCreateConfig returns VA_STATUS_ERROR_ATTR_NOT_SUPPORTED.

feiwan1, 2018-10-16:
1. cmdline:
gst-launch-1.0 videotestsrc num-buffers=100 ! vaapih265enc tune=3 rate-control=cbr ! fakesink
(Note: this cmdline need a patch for gst-vaapi, please feel free to ping me if you need the patch for debug).

2. error log:
root@icl_vpg_kernel:~# gst-launch-1.0 videotestsrc num-buffers=100 ! vaapih265enc tune=3 rate-control=cbr ! fakesink
Setting pipeline to PAUSED ...
Pipeline is PREROLLING ...
Got context from element 'vaapiencodeh265-0': gst.vaapi.Display=context, gst.vaapi.Display=(GstVaapiDisplay)"\(GstVaapiDisplayDRM\)\ vaapidisplaydrm1";
0:00:00.035357481 17100      0x1e55850 ERROR                  vaapi 0:00:00.035389684 17100      0x1e55850 ERROR                  vaapi gstvaapiencoder.c:997:gst_vaapi_encoder_reconfigure_internal: failed to update VA context

3. bt information:
(gdb) bt
#0  MediaLibvaCaps::CreateEncConfig (this=this@entry=0x785a00, profileTableIdx=29, entrypoint=entrypoint@entry=VAEntrypointEncSliceLP, attribList=attribList@entry=0x7fffeb87d880, numAttribs=n
umAttribs@entry=3, configId=0x7fffe4003a58) at /root/build/media-driver/media_driver/linux/common/ddi/media_libva_caps.cpp:1674
#1  0x00007fffec2a5cfc in MediaLibvaCaps::CreateConfig (this=0x785a00, profile=VAProfileHEVCMain, entrypoint=VAEntrypointEncSliceLP, attribList=0x7fffeb87d880, numAttribs=3, configId=<optimized out>) at /
root/build/media-driver/media_driver/linux/common/ddi/media_libva_caps.cpp:1867
#2  0x00007ffff356eba9 in vaCreateConfig (dpy=0x72aa60, profile=VAProfileHEVCMain, entrypoint=VAEntrypointEncSliceLP, attrib_list=attrib_list@entry=0x7fffeb87d880, num_attribs=num_attribs@entry=3, config_
id=config_id@entry=0x7fffe4003a58) at va.c:891

3. variable in MediaLibvaCaps::CreateEncConfig:
(gdb) p m_profileEntryTbl[29]
$47 = {m_profile = VAProfileHEVCMain, m_entrypoint = VAEntrypointEncSliceLP, m_attributes = 0x787390, m_configStartIdx = 73, m_configNum = 9}

(gdb) p m_encConfigs
$41 = std::vector of length 125, capacity 128 = {16, 2, 4, 130, 132, 64, 8, 16, 2,
  4, 130, 132, 64, 8, 16, 2, 4, 130, 132, 64, 8, 16, 16, 16, 16, 2, 4, 130,
  132, 16, 2, 4, 130, 132, 16, 2, 4, 130, 132, 16, 2, 4, 16, 2, 4, 1, 16, 2,
  514, 4, 516, 130, 642, 132, 644, 64, 576, 8, 520, 16, 16, 2, 514, 4, 516,
  130, 642, 132, 644, 64, 576, 8, 520, 16, 130, 642, 132, 644, 64, 576, 8,
  520, 16, 130, 642, 132, 644, 64, 576, 8, 520, 16, 130, 642, 132, 644, 64,
  576, 8, 520, 16, 130, 642, 132, 644, 64, 576, 8, 520, 16, 2, 4, 16, 2, 4,
  16, 2, 4, 16, 2, 4, 16, 2, 4, 1}

4. VA_STATUS_ERROR_ATTR_NOT_SUPPORTED returned in else branch:
1682│     if (j < (configNum + startIdx))
1683│     {
1684│         *configId = j + DDI_CODEC_GEN_CONFIG_ATTRIBUTES_ENC_BASE;
1685│         return VA_STATUS_SUCCESS;
1686│     }
1687│     else
1688│     {
1689│         *configId = 0xFFFFFFFF;
1690│         return VA_STATUS_ERROR_ATTR_NOT_SUPPORTED;
1691│     }

XinfengZhang, 2018-10-31:
@wangyan42164 please help to check

wangyan42164, 2018-10-31:
@feiwan1 @XinfengZhang I submit it one RFC patch for review. I think it is similar with the previous closed issue https://github.com/intel/media-driver/issues/8.
Thanks.
Yan Wang

wangyan42164, 2018-10-31:
@feiwan1 Sorry for missing "break" in the previous version. I updated it. Could you please try it again? Thanks.

wangyan42164, 2018-11-01:
@feiwan1 Could you please provide back trace for analysis too? Thanks.

feiwan1, 2019-02-02:
@wangyan42164 , any update for this issue?

feiwan1, 2019-02-19:
@XinfengZhang @wangyan42164 @Sherry-Lin 
I used a new version of kernel/guc/huc from Emma, but still have gpu hang issue.

cmdline(this cmdline need a patch on gst-vaapi, see attatchment):
root@icl_1604:~/build/gstreamer-vaapi# gst-launch-1.0 videotestsrc num-buffers=100 ! vaapih265enc tune=low-power quality-level=4 ! filesink location=test.h265

root@icl_1604:~# dmesg 
[63955.743664] i915 0000:00:02.0: Resetting vcs0 after gpu hang

[0001-encoder-h265-Add-support-for-low-power-high-performa.patch.zip](https://github.com/intel/media-driver/files/2878070/0001-encoder-h265-Add-support-for-low-power-high-performa.patch.zip)

zjamy, 2019-04-08:
When will https://github.com/intel/media-driver/pull/391 be merged? The gstreamer-vaapi test pass with patch#391.

PR #391 (2019-04-11): [RFC] Add VA_RC_MB flag for VAEntrypointEncSliceLP+HEVC encoding.
Fixes #361.
commit f74f1c38 (2019-04-11): Add VA_RC_MB flag for VAEntrypointEncSliceLP+HEVC encoding.

When VA_RC_CBR or VA_RC_VBR is used for VAEntrypointEncSliceLP+HEVC
encoding, VA_RC_MB must be set too.
Fixes #361.
closed 2019-04-11 by intel-mediadev