intel/media-driver #362: MediaDecodeDdiTest.DecodeHEVCLong fails during make

andreasunterhuber, 2018-10-16:
Running current GIT `media-driver` and current GIT `gmmlib` on `i7-8700K CPU` as

1. **cmake** command
`cmake ../media-driver -DMEDIA_VERSION="2.0.0" -DBUILD_ALONG_WITH_CMRTLIB=1 -DBS_DIR_GMMLIB=`pwd`/../gmmlib/Source/GmmLib/ -DBS_DIR_COMMON=`pwd`/../gmmlib/Source/Common/ -DBS_DIR_INC=`pwd`/../gmmlib/Source/inc/ -DBS_DIR_MEDIA=`pwd`/../media-driver -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_INSTALL_LIBDIR=/usr/lib64 -DINSTALL_DRIVER_SYSCONF=OFF -DLIBVA_DRIVERS_PATH=/usr/lib64/dri`

2. **make** command
`make -j$(nproc)`

3. Gives **results**
```
make -j$(nproc)
[  2%] Built target drm_mock
[  2%] Built target libgtest
[  5%] Built target igfxcmrt
[ 11%] Built target devult
[100%] Built target iHD_drv_video_OBJ
[100%] Built target iHD_drv_video_STATIC
[100%] Built target iHD_drv_video
Running devult...
[==========] Running 43 tests from 13 test cases.
[----------] Global test environment set-up.
[----------] 1 test from MediaCapsDdiTest
[ RUN      ] MediaCapsDdiTest.DecodeEncodeProfile
[       OK ] MediaCapsDdiTest.DecodeEncodeProfile (12 ms)
[----------] 1 test from MediaCapsDdiTest (12 ms total)

[----------] 2 tests from MediaDecodeDdiTest
[ RUN      ] MediaDecodeDdiTest.DecodeHEVCLong
/root/installer/media-driver/media_driver/linux/ult/ult_app/ddi_test_decode.cpp:78: Failure
      Expected: 0x00000000
      Which is: 0
To be equal to: ret
      Which is: 2
Platform = SKL, Failed function = m_driverLoader.m_ctx.vtable->vaCreateSurfaces2

/root/installer/media-driver/media_driver/linux/ult/ult_app/ddi_test_decode.cpp:83: Failure
      Expected: 0x00000000
      Which is: 0
To be equal to: ret
      Which is: 6
Platform = SKL, Failed function = m_driverLoader.m_ctx.vtable->vaCreateContext

/root/installer/media-driver/media_driver/linux/ult/ult_app/ddi_test_decode.cpp:90: Failure
      Expected: 0x00000000
      Which is: 0
To be equal to: ret
      Which is: 6
Platform = SKL, Failed function = m_driverLoader.m_ctx.vtable->vaBeginPicture

/root/installer/media-driver/media_driver/linux/ult/ult_app/ddi_test_decode.cpp:98: Failure
      Expected: 0x00000000
      Which is: 0
To be equal to: ret
      Which is: 5
Platform = SKL, Failed function = m_driverLoader.m_ctx.vtable->vaCreateBuffer

/root/installer/media-driver/media_driver/linux/ult/ult_app/ddi_test_decode.cpp:98: Failure
      Expected: 0x00000000
      Which is: 0
To be equal to: ret
      Which is: 5
Platform = SKL, Failed function = m_driverLoader.m_ctx.vtable->vaCreateBuffer

/root/installer/media-driver/media_driver/linux/ult/ult_app/ddi_test_decode.cpp:98: Failure
      Expected: 0x00000000
      Which is: 0
To be equal to: ret
      Which is: 5
Platform = SKL, Failed function = m_driverLoader.m_ctx.vtable->vaCreateBuffer

/root/installer/media-driver/media_driver/linux/ult/ult_app/ddi_test_decode.cpp:108: Failure
      Expected: 0x00000000
      Which is: 0
To be equal to: ret
      Which is: 7
Platform = SKL, Failed function = m_driverLoader.m_ctx.vtable->vaRenderPicture

/root/installer/media-driver/media_driver/linux/ult/ult_app/ddi_test_decode.cpp:108: Failure
      Expected: 0x00000000
      Which is: 0
To be equal to: ret
      Which is: 7
Platform = SKL, Failed function = m_driverLoader.m_ctx.vtable->vaRenderPicture

/root/installer/media-driver/media_driver/linux/ult/ult_app/ddi_test_decode.cpp:108: Failure
      Expected: 0x00000000
      Which is: 0
To be equal to: ret
      Which is: 7
Platform = SKL, Failed function = m_driverLoader.m_ctx.vtable->vaRenderPicture

/root/installer/media-driver/media_driver/linux/ult/ult_app/ddi_test_decode.cpp:113: Failure
      Expected: 0x00000000
      Which is: 0
To be equal to: ret
      Which is: 5
Platform = SKL, Failed function = m_driverLoader.m_ctx.vtable->vaEndPicture
```

Possible to avoid this error during make with `-DBYPASS_MEDIA_ULT=yes` (as additional cmake param) - but anyway... the media-driver seems to crash: 

Test-command `/opt/intel/mediasdk/share/mfx/samples/sample_multi_transcode -i::h264 ~/test.h264 -o::h264 ~/test_out.h264 -hw` gives: 

```
Multi Transcoding Sample Version 8.3.26.

libva info: VA-API version 1.3.0
libva info: va_getDriverName() returns 0
libva info: User requested driver 'iHD'
libva info: Trying to open /usr/lib64/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_3
libva info: va_openDriver() returns 0
Pipeline surfaces number (DecPool): 7

[ERROR], sts=MFX_ERR_DEVICE_FAILED(-17), Init, m_pmfxDEC->Init failed at /root/installer/MediaSDK/samples/sample_multi_transcode/src/pipeline_transcode.cpp:3823

[ERROR], sts=MFX_ERR_DEVICE_FAILED(-17), Init, pThreadPipeline->pPipeline->Init failed at /root/installer/MediaSDK/samples/sample_multi_transcode/src/sample_multi_transcode.cpp:299

[ERROR], sts=MFX_ERR_DEVICE_FAILED(-17), main, transcode.Init failed at /root/installer/MediaSDK/samples/sample_multi_transcode/src/sample_multi_transcode.cpp:881
```

andreasunterhuber, 2018-10-16:
Changing gmmlib branch to `intel-gmmlib-18.3` solves the problem. Some newer commit in **master** branch seems to create problems and should be investigated. Thanks in advance.

chivakker, 2019-07-02:
Problem still happens with ToT and gmmlib 9.0.0.  Can tests be confirmed to work?

wangyan-intel, 2019-07-02:
@chivakker I have not i7-8700K in my hand. I tried KBL and ICL with the latest media driver and gmmlib. No this issue. Could you please double check it? Thanks.

wangyan-intel, 2019-08-29:
Keep it one week. If no more response, I will close it.

wangyan-intel, 2020-01-19:
No response. Close it.

closed 2020-01-19 by wangyan-intel