intel/media-driver #363: [regression] over 1600 failures in libva-utils conformance tests

uartie, 2018-10-16:
There are over 1600 new test regressions in libva-utils conformance test suite (test_va_api) introduced since:

```
commit 9fbf6c9ad5ed226e91b97e2ee6a19272c6e5a2f3
Author: shiqiang_Yu <shiqiang.yu@intel.com>
Date:   Mon Oct 15 09:39:47 2018 +0800

    [Decode] Linux DDI code cleanup DDI(CL#807586)
    
    Linux DDI code cleanup DDI
    
    Change-Id: I53024acb5dcf1e4b3f6b1e3d9438cc1064c900b7
```

XinfengZhang, 2018-10-17:
thanks reporting, will follow up immediately

shiqiangyu, 2018-10-19:
Here is a patch for fix these new regressions. uartie, could you help me verify the patch within you env. Thanks.

[remove_check_nullptr_for_zero_size_array.zip](https://github.com/intel/media-driver/files/2494136/remove_check_nullptr_for_zero_size_array.zip)

uartie, 2018-10-19:
@shiqiangyu please build and install https://github.com/intel/libva-utils and run `test_va_api` to validate your patch.  I don't have bandwidth to sling around patch archives.

uartie, 2018-10-19:
`test_va_api` should take less than a minute to execute.  Would be good if all devs incorporated this test suite into their workflow before submitting patches.

uartie, 2018-10-26:
3c55dfe3061e49a3507d50cf60dafe12583b506c reverts the regressing commit. Please reference issue numbers (e.g. #363) in commit messages. GitHub will automatically create link between issue and commit and makes it easier to see when activity occurs for specific issue.

closed 2019-08-29 by XinfengZhang