intel/media-driver #366: [regression] nearly all ffmpeg-qsv and many gst-msdk pipelines fail

uartie, 2018-10-17:
First bad commit is:

```
commit 7aa283c2f54cba5c1295bbd1c100b720dd430205 (HEAD -> master, origin/master, origin/HEAD)
Author: fen_(molly)__hu <molly.hu@intel.com>
Date:   Wed Oct 17 10:08:08 2018 +0800

    [VP] Fixing compostion vp event notification more than once for one VPBLIT(CL#808344)
    
    When render do compostion may sumbit cmdbuffer more than once for one VPBLIT. It need  set the last sumbition's kmd mediaFrameTracking as true, Then the KMD just notify VP event once for one VPBLIT.
    
    Change-Id: Ib0d9348965b0cd7249a862f04a94e52b296ec40a
```

Nearly every encode, decode and vpp pipeline for ffmpeg-qsv no longer work.

For example:

```
$ ffmpeg -hwaccel qsv -hwaccel_device /dev/dri/renderD128 -v verbose -c:v h264_qsv \
 -i ./AUD_MW_E.264 -vf 'hwdownload,format=nv12' -pix_fmt yuv420p -f rawvideo \
 -vsync passthrough -vframes 4 -y output.yuv

[2018-10-17 16:36:54] ffmpeg version 4.0.git Copyright (c) 2000-2018 the FFmpeg developers
[2018-10-17 16:36:54]   built with gcc 8 (GCC)
[2018-10-17 16:36:54]   configuration: --prefix=/home/uaeoff/Work/workspace/media/install --disable-static --enable-shared --enable-libdrm --enable-vaapi --enable-libmfx --disable-amf --disable-audiotoolbox --disable-cuda --disable-cuda-sdk --disable-cuvid --disable-d3d11va --disable-dxva2 --disable-libnpp --disable-mmal --disable-nvdec --disable-nvenc --disable-omx --disable-omx-rpi --disable-rkmpp --disable-v4l2-m2m --disable-vdpau --disable-videotoolbox --enable-nonfree --enable-gpl
[2018-10-17 16:36:54]   libavutil      56. 19.101 / 56. 19.101
[2018-10-17 16:36:54]   libavcodec     58. 33.100 / 58. 33.100
[2018-10-17 16:36:54]   libavformat    58. 18.104 / 58. 18.104
[2018-10-17 16:36:54]   libavdevice    58.  4.105 / 58.  4.105
[2018-10-17 16:36:54]   libavfilter     7. 33.101 /  7. 33.101
[2018-10-17 16:36:54]   libswscale      5.  2.100 /  5.  2.100
[2018-10-17 16:36:54]   libswresample   3.  2.100 /  3.  2.100
[2018-10-17 16:36:54]   libpostproc    55.  2.100 / 55.  2.100
[2018-10-17 16:36:54] [h264 @ 0x157b200] Reinit context to 176x144, pix_fmt: yuv420p
[2018-10-17 16:36:54] Input #0, h264, from '/home/uaeoff/Work/workspace/media/src/otc-media/assets/avc/AUD_MW_E.264':
[2018-10-17 16:36:54]   Duration: N/A, bitrate: N/A
[2018-10-17 16:36:54]     Stream #0:0: Video: h264 (Constrained Baseline), 1 reference frame, yuv420p(progressive, left), 176x144, 25 fps, 25 tbr, 1200k tbn, 50 tbc
[2018-10-17 16:36:54] [AVBSFContext @ 0x16251c0] The input looks like it is Annex B already
[2018-10-17 16:36:54] Stream mapping:
[2018-10-17 16:36:54]   Stream #0:0 -> #0:0 (h264 (h264_qsv) -> rawvideo (native))
[2018-10-17 16:36:54] Press [q] to stop, [?] for help
[2018-10-17 16:36:54] [AVHWDeviceContext @ 0x1622940] Opened VA display via X11 display :0.
[2018-10-17 16:36:54] [AVHWDeviceContext @ 0x1622940] libva: VA-API version 1.3.0
[2018-10-17 16:36:54] [AVHWDeviceContext @ 0x1622940] libva: va_getDriverName() returns 0
[2018-10-17 16:36:54] [AVHWDeviceContext @ 0x1622940] libva: User requested driver 'iHD'
[2018-10-17 16:36:54] [AVHWDeviceContext @ 0x1622940] libva: Trying to open /home/uaeoff/Work/workspace/media/install/lib/dri/iHD_drv_video.so
[2018-10-17 16:36:54] [AVHWDeviceContext @ 0x1622940] libva: Found init function __vaDriverInit_1_3
[2018-10-17 16:36:54] [AVHWDeviceContext @ 0x1622940] libva: va_openDriver() returns 0
[2018-10-17 16:36:54] [AVHWDeviceContext @ 0x1622940] Initialised VAAPI connection: version 1.3
[2018-10-17 16:36:54] [AVHWDeviceContext @ 0x1622940] VAAPI driver: Intel iHD driver - 1.0.0.
[2018-10-17 16:36:54] [AVHWDeviceContext @ 0x1622940] Driver not found in known nonstandard list, using standard behaviour.
[2018-10-17 16:36:54] [AVHWDeviceContext @ 0x1624d00] Initialize MFX session: API version is 1.28, implementation version is 1.28
[2018-10-17 16:36:54] [AVHWDeviceContext @ 0x1624d00] MFX compile/runtime API: 1.28/1.28
[2018-10-17 16:36:54] [AVHWDeviceContext @ 0x15c2740] VAAPI driver: Intel iHD driver - 1.0.0.
[2018-10-17 16:36:54] [AVHWDeviceContext @ 0x15c2740] Driver not found in known nonstandard list, using standard behaviour.
[2018-10-17 16:36:54] [graph 0 input from stream 0:0 @ 0x16d45c0] w:176 h:144 pixfmt:qsv tb:1/1200000 fr:25/1 sar:0/1 sws_param:flags=2
[2018-10-17 16:36:54] [auto_scaler_0 @ 0x1a499c0] w:iw h:ih flags:'bicubic' interl:0
[2018-10-17 16:36:54] [format @ 0x1a46580] auto-inserting filter 'auto_scaler_0' between the filter 'Parsed_format_1' and the filter 'format'
[2018-10-17 16:36:54] [auto_scaler_0 @ 0x1a499c0] w:176 h:144 fmt:nv12 sar:0/1 -> w:176 h:144 fmt:yuv420p sar:0/1 flags:0x4
[2018-10-17 16:36:54] [AVHWFramesContext @ 0x1621280] Error synchronizing the operation: -21
[2018-10-17 16:36:54] [hwdownload @ 0x16d96c0] Failed to download frame: -1313558101.
[2018-10-17 16:36:54] Error while filtering: Unknown error occurred
[2018-10-17 16:36:54] Failed to inject frame into filter network: Unknown error occurred
[2018-10-17 16:36:54] Error while processing the decoded data for stream #0:0
[2018-10-17 16:36:54] [AVIOContext @ 0x15c7300] Statistics: 0 seeks, 0 writeouts
[2018-10-17 16:36:54] [AVIOContext @ 0x15824c0] Statistics: 54828 bytes read, 0 seeks
[2018-10-17 16:36:54] Conversion failed!
```

Also, this broke many vpp pipelines in gst-msdk, for example:

```
$ LIBVA_DRIVER_NAME=iHD gst-launch-1.0 -vf videotestsrc num-buffers=10 \
 ! video/x-raw,format=NV12,width=1920,height=1080 \
 ! msdkvpp hardware=true brightness=100.0 ! fakesink

Setting pipeline to PAUSED ...
libva info: VA-API version 1.3.0
libva info: va_getDriverName() returns 0
libva info: User requested driver 'iHD'
libva info: Trying to open /home/uaeoff/Work/workspace/media/install/lib/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_3
libva info: va_openDriver() returns 0
Pipeline is PREROLLING ...
Got context from element 'msdkvpp0': gst.msdk.Context=context, gst.msdk.Context=(GstMsdkContext)"\(GstMsdkContext\)\ msdkcontext0";
/GstPipeline:pipeline0/GstVideoTestSrc:videotestsrc0.GstPad:src: caps = video/x-raw, format=(string)NV12, width=(int)1920, height=(int)1080, framerate=(fraction)30/1, multiview-mode=(string)mono, interlace-mode=(string)progressive, pixel-aspect-ratio=(fraction)1/1
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:src: caps = video/x-raw, format=(string)NV12, width=(int)1920, height=(int)1080, framerate=(fraction)30/1, multiview-mode=(string)mono, interlace-mode=(string)progressive, pixel-aspect-ratio=(fraction)1/1
/GstPipeline:pipeline0/GstMsdkVPP:msdkvpp0.GstPad:src: caps = video/x-raw, format=(string)BGRA, width=(int)1920, height=(int)1080, framerate=(fraction)30/1, pixel-aspect-ratio=(fraction)1/1, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, interlace-mode=(string)progressive
/GstPipeline:pipeline0/GstFakeSink:fakesink0.GstPad:sink: caps = video/x-raw, format=(string)BGRA, width=(int)1920, height=(int)1080, framerate=(fraction)30/1, pixel-aspect-ratio=(fraction)1/1, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, interlace-mode=(string)progressive
/GstPipeline:pipeline0/GstMsdkVPP:msdkvpp0.GstPad:sink: caps = video/x-raw, format=(string)NV12, width=(int)1920, height=(int)1080, framerate=(fraction)30/1, multiview-mode=(string)mono, interlace-mode=(string)progressive, pixel-aspect-ratio=(fraction)1/1
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:sink: caps = video/x-raw, format=(string)NV12, width=(int)1920, height=(int)1080, framerate=(fraction)30/1, multiview-mode=(string)mono, interlace-mode=(string)progressive, pixel-aspect-ratio=(fraction)1/1
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
0:00:00.100991523 22810      0x15aa2d0 ERROR                msdkvpp gstmsdkvpp.c:717:gst_msdkvpp_transform:<msdkvpp0> MSDK Failed to do VPP
ERROR: from element /GstPipeline:pipeline0/GstVideoTestSrc:videotestsrc0: Internal data stream error.
Additional debug info:
gstbasesrc.c(3064): gst_base_src_loop (): /GstPipeline:pipeline0/GstVideoTestSrc:videotestsrc0:
streaming stopped, reason error (-5)
Execution ended after 0:00:00.008054162
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...
```

uartie, 2018-10-17:
Also related: #365

fenhu, 2018-10-18:
Hi uartie.

 I back out my CL which case this issue  to OpenSource  by change https://gerrit-gfx.intel.com/#/c/505327/

Please help to try it.

Thank you
Molly

uartie, 2018-10-18:
@fenhu sorry, but I don't have access to your gerrit system.  Please revert it here in GitHub.

Sherry-Lin, 2018-10-18:
reverted in commit#7bd497ee0493a8314155d18233d2e124e663b304

eero-t, 2019-01-16:
Why this is still open?  Is there still something to fix?

XinfengZhang, 2019-08-29:
add fixes keyword for future reverting should be better to trace the issue

closed 2019-08-29 by XinfengZhang