intel/media-driver #367: Incorrect intel-media.sh install path within buildroot

eero-t, 2018-10-18:
Use-case:
* Use (Gnome) JHBuild for building Intel Media stack
* "/opt/install/Custom_4026" is a symlink to "/opt/install" directory
* JHBuild sets buildroot to /opt/install/Custom_4026/root-media-driver
* and gives CMake "-DCMAKE_INSTALL_PREFIX=/opt/install/Custom_4026" option
* "make install"

Expected outcome:
* intel-media.sh goes to:
  /opt/install/Custom_4026/root-media-driver/opt/install/Custom_4026/etc/profile.d/intel-media.sh

Actual outcome:
* intel-media.sh goes to:
  /opt/install/Custom_4026/root-media-driver/etc/opt/install/Custom_4026/profile.d/intel-media.sh

I.e. to /etc/$PREFIX/profile.d/ path instead of correct $PREFIX/etc/profile.d/ path.

Media-SDK doesn't have the same problem, and neither have none of the other files in media-driver.

Somehow CMake digs out the whole path, including buildroot part (Autotools doesn't), maybe that somehow confuses how media-driver determines correct path to /etc?

yangnin, 2018-11-05:
Hi, 
media driver uses CMAKE_INSTALL_FULL_SYSCONFDIR as intel-media.sh destination.
Please reference
https://cmake.org/cmake/help/v3.4/module/GNUInstallDirs.html

eero-t, 2018-11-05:
Wow.  I had no idea that Filesystem Hierarchy Standard (FHS) requires non-standard handling for "etc" directory when it's for the /opt/ directory hierarchy:
http://refspecs.linuxfoundation.org/FHS_3.0/fhs/ch03s07.html#etcoptConfigurationFilesForOpt

I'm not sure moving "etc" out of build target should be taken by the build system (CMake) though, as there's no guarantee that a directory outside of build target directory is writable by the builder (it could be chrooted / containerized, or just not have write access there).

But media-driver itself works as expected, this is not a bug.

closed 2018-11-05 by eero-t