intel/media-driver #368: Example programs installed to incorrect directory

eero-t, 2018-10-18:
Media driver installs example programs like "sample_multi_transcode" under $PREFIX/share/ directory, which is reserved for non-architecture specific files:
https://refspecs.linuxfoundation.org/FHS_3.0/fhs/ch04s11.html

Examples in general would be better in source form, and then they would should be installed under share/, e.g. to share/<package-name>/examples/.

Binaries should be under *bin/ or lib*/ directory hierarchies:
https://refspecs.linuxfoundation.org/FHS_3.0/fhs/ch04s04.html
https://refspecs.linuxfoundation.org/FHS_3.0/fhs/ch04s06.html
https://refspecs.linuxfoundation.org/FHS_3.0/fhs/ch04s07.html

Either directly in bin/ directory (if those examples are thought to be useful in general), or e.g. in /usr/lib/media-driver/examples/.

ljia5, 2020-06-04:
should not open source issue. We cannot get media package. Please transfer to msdk?

eero-t, 2020-06-04:
Filed issue under MSDK: https://github.com/Intel-Media-SDK/MediaSDK/issues/2148

closed 2020-06-04 by eero-t