intel/media-driver #369: Hardward Acceleration (QSV) Causes Problems when decoding videos.

ChrisDoesEverything, 2018-10-18:
I've tried everything and couldn't find a REAL solution to this problem.

Proof 1:
I've installed a fresh copy of Windows and Chrome web browser, when I view some videos I get a black screen for a fraction of a section on random intervals. In Chrome hardware acceleration setting is enable by default and when disabling it, it seems to work normally.

Proof 2:
I used a screen recording program called OBS and I've recorded the screen for a few seconds with the SQV hardware encoding enabled. The video is recorded with no problem but while playing it back, artifacts appear on the video such as: https://i.imgur.com/Xr0qpRl.png
While recording without QSV encoding but software encoding it records well and plays it back well which is a little bit weird but okay.

Also the following event log appears on Windows while the black screen appears:

> The application-specific permission settings do not grant Local Activation permission for the COM Server application with CLSID 
> {D63B10C5-BB46-4990-A94F-E40B9D520160}
>  and APPID 
> {9CA88EE3-ACB7-47C8-AFC4-AB702511C276}
>  to the user NT AUTHORITY\LOCAL SERVICE SID (S-1-5-19) from address LocalHost (Using LRPC) running in the application container Unavailable SID (Unavailable). This security permission can be modified using the Component Services administrative tool.
>

XinfengZhang, 2020-06-07:
two issue reported, 1. is chromium issue , suppose it is VP9 or VP8, need to verify whether the issue still there, we have several fix related with it. 2. OBS/QSV encode issue.

XinfengZhang, 2022-09-05:
@KrisKnez , is there some new finding, looks it is too old , and lost track, I would like to close it, if you have any concern , please re-open it directly. thanks

closed 2022-09-05 by XinfengZhang