intel/media-driver #372: [regression] ffmpeg-vaapi [M]JPEG encode quality drop

uartie, 2018-10-19:
With ffmpeg-vaapi, [M]JPEG encoding quality dropped significantly for all input YUV files and quality settings since:

```
commit 9fbf6c9ad5ed226e91b97e2ee6a19272c6e5a2f3
Author: shiqiang_Yu <shiqiang.yu@intel.com>
Date:   Mon Oct 15 09:39:47 2018 +0800

    [Decode] Linux DDI code cleanup DDI(CL#807586)
    
    Linux DDI code cleanup DDI
    
    Change-Id: I53024acb5dcf1e4b3f6b1e3d9438cc1064c900b7
```

For example:
```
$ LIBVA_DRIVER_NAME=iHD ffmpeg -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -v debug \
  -f rawvideo -pix_fmt yuv420p -s:v 1280x720 \
  -i ./stockholm_1280x720_604_1280x720_I420.yuv \
  -vf 'format=nv12,hwupload' -c:v mjpeg_vaapi -global_quality 100 \
  -vframes 10 -y ./stockholm-100.mjpeg

$ LIBVA_DRIVER_NAME=iHD ffmpeg -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -v debug \
  -i ./stockholm-100.mjpeg -pix_fmt yuv420p -f rawvideo \
  -vsync passthrough -vframes 10 -y ./stockholm-100-1280x720-I420.yuv

# run PSNR metric between ./stockholm_1280x720_604_1280x720_I420.yuv and ./stockholm-100-1280x720-I420.yuv
```

PSNR for above example **before** regression commit:
```
MIN_PSNR_Y = 61.37220901965478
MIN_PSNR_U = 64.03241128896165
MIN_PSNR_V = 63.96182824127429
AVG_PSNR_Y = 61.48656699371482
AVG_PSNR_U = 64.23875757635905
AVG_PSNR_V = 66.696503932905
```

PSNR for above example **after** regression commit:
```
MIN_PSNR_Y = 30.311048523575124
MIN_PSNR_U = 46.570752174165229
MIN_PSNR_V = 47.33517578328653
AVG_PSNR_Y = 31.051680273070975
AVG_PSNR_U = 50.08017694983134
AVG_PSNR_V = 53.418125584418235
```

shiqiangyu, 2018-10-22:
@uartie, what's the command for comparing the psnr? And could you attach the yuv file for reproducing the psnr degradation issue? Thanks.

shiqiangyu, 2018-10-22:
I reproduced this issues locally, it shared the same root cause with the 1600 libva-utils conformance test fails. With the patch I attached previously, psnr degradation issue disappeared. And the patch is under reviewing, will upstream soon.

uartie, 2018-10-24:
FYI, it looks like codecs in ffmpeg-vaapi decode is falling back to SW decode with commit 9fbf6c9ad5ed226e.  So the problem is worse than just mjpeg.  I have updated our tests (that I shared with you) to detect this sw fallback and fail.  Please verify patch fixes this too.

shiqiangyu, 2018-10-25:
@uartie Sure, will check this, but now I have no idea how to detect whether it's using SW decode or not from the output. How about check in the patch first? Will follow up if another root cause for the decoder fallback issue.

uartie, 2018-10-25:
@shiqiangyu yes, please submit the patch ASAP.

For this particular regression, FFmpeg will output the following when it fails to init hwaccel:
```
[mjpeg @ 0x1551580] Failed to create decode configuration: 18 (invalid parameter).
[mjpeg @ 0x1551580] Failed setup for format vaapi_vld: hwaccel initialisation returned error.
```
There are many other potential points of failure and arbitrarily different error messages in ffmpeg hwaccel init phase.  FFmpeg automatically switches to SW mode when any of these errors occur.  This makes it difficult to "auto-detect" when this happens.  It would be nice if FFmpeg allowed command-line disable of SW fallback so it can be detected easier when hwaccel fails.

uartie, 2018-10-26:
3c55dfe3061e49a3507d50cf60dafe12583b506c reverts the regressing commit. Please reference issue numbers (e.g. #372) in commit messages. GitHub will automatically create link between issue and commit and makes it easier to see when activity occurs for specific issue.

eero-t, 2019-01-16:
`3c55dfe reverts the regressing commit. Please reference issue numbers (e.g. #372) in commit messages.`
Is there still something to fix, or why this bug is still open?

XinfengZhang, 2019-08-29:
@xhaihao is working on a similiar issue , media-driver scale the qmatrix by the quality setting but application already scale it

XinfengZhang, 2019-08-29:
this issue can be closed, because it is caused by regression. but still another jpeg encode quality issue #281 will track the issue I mentioned.

closed 2019-08-29 by XinfengZhang