intel/media-driver #373: [regression] ffmpeg-qsv [M]JPEG encode commands fail (conversion failed!)

uartie, 2018-10-19:
For ffmpeg-qsv, all [M]JPEG encode pipeline commands fail since:

```
commit 9fbf6c9ad5ed226e91b97e2ee6a19272c6e5a2f3
Author: shiqiang_Yu <shiqiang.yu@intel.com>
Date:   Mon Oct 15 09:39:47 2018 +0800

    [Decode] Linux DDI code cleanup DDI(CL#807586)
    
    Linux DDI code cleanup DDI
    
    Change-Id: I53024acb5dcf1e4b3f6b1e3d9438cc1064c900b7
```

```
$ ffmpeg -init_hw_device qsv=qsv:hw -hwaccel qsv -filter_hw_device qsv -v verbose \
  -f rawvideo -pix_fmt yuv420p -s:v 1280x720 -i ./input.yuv \
  -vf 'format=nv12,hwupload=extra_hw_frames=64' \
  -c:v mjpeg_qsv -global_quality 100 -vframes 10 \
  -y output.mjpeg

ffmpeg version 4.0.git Copyright (c) 2000-2018 the FFmpeg developers
  built with gcc 8 (GCC)
  configuration: --prefix=/home/uaeoff/Work/workspace/media/install --disable-static --enable-shared --enable-libdrm --enable-vaapi --enable-libmfx --disable-amf --disable-audiotoolbox --disable-cuda --disable-cuda-sdk --disable-cuvid --disable-d3d11va --disable-dxva2 --disable-libnpp --disable-mmal --disable-nvdec --disable-nvenc --disable-omx --disable-omx-rpi --disable-rkmpp --disable-v4l2-m2m --disable-vdpau --disable-videotoolbox --enable-nonfree --enable-gpl
  libavutil      56. 19.101 / 56. 19.101
  libavcodec     58. 33.101 / 58. 33.101
  libavformat    58. 19.100 / 58. 19.100
  libavdevice    58.  4.105 / 58.  4.105
  libavfilter     7. 34.100 /  7. 34.100
  libswscale      5.  2.100 /  5.  2.100
  libswresample   3.  2.100 /  3.  2.100
  libpostproc    55.  2.100 / 55.  2.100
[AVHWDeviceContext @ 0x1cda900] Opened VA display via X11 display :0.
[AVHWDeviceContext @ 0x1cda900] libva: VA-API version 1.3.0
[AVHWDeviceContext @ 0x1cda900] libva: va_getDriverName() returns 0
[AVHWDeviceContext @ 0x1cda900] libva: Trying to open /home/uaeoff/Work/workspace/media/install/lib/dri/i965_drv_video.so
[AVHWDeviceContext @ 0x1cda900] libva: Found init function __vaDriverInit_1_3
[AVHWDeviceContext @ 0x1cda900] libva: va_openDriver() returns 0
[AVHWDeviceContext @ 0x1cda900] Initialised VAAPI connection: version 1.3
[AVHWDeviceContext @ 0x1cda900] VAAPI driver: Intel i965 driver for Intel(R) Kaby Lake - 2.2.1.pre1 (2.2.1.pre1).
[AVHWDeviceContext @ 0x1cda900] Driver not found in known nonstandard list, using standard behaviour.
[AVHWDeviceContext @ 0x1cda4c0] Initialize MFX session: API version is 1.28, implementation version is 1.28
[AVHWDeviceContext @ 0x1cda4c0] MFX compile/runtime API: 1.28/1.28
[rawvideo @ 0x1d08700] Estimating duration from bitrate, this may be inaccurate
Input #0, rawvideo, from '/home/uaeoff/Work/workspace/media/src/otc-media/assets/yuv/input.yuv':
  Duration: 00:00:24.16, start: 0.000000, bitrate: 276480 kb/s
    Stream #0:0: Video: rawvideo, 1 reference frame (I420 / 0x30323449), yuv420p, 1280x720, 276480 kb/s, 25 tbr, 25 tbn, 25 tbc
Stream mapping:
  Stream #0:0 -> #0:0 (rawvideo (native) -> mjpeg (mjpeg_qsv))
Press [q] to stop, [?] for help
[graph 0 input from stream 0:0 @ 0x1d27d80] w:1280 h:720 pixfmt:yuv420p tb:1/25 fr:25/1 sar:0/1 sws_param:flags=2
[auto_scaler_0 @ 0x1d2b280] w:iw h:ih flags:'bicubic' interl:0
[Parsed_format_0 @ 0x1d254c0] auto-inserting filter 'auto_scaler_0' between the filter 'graph 0 input from stream 0:0' and the filter 'Parsed_format_0'
[auto_scaler_0 @ 0x1d2b280] w:1280 h:720 fmt:yuv420p sar:0/1 -> w:1280 h:720 fmt:nv12 sar:0/1 flags:0x4
[AVHWDeviceContext @ 0x1d27600] VAAPI driver: Intel i965 driver for Intel(R) Kaby Lake - 2.2.1.pre1 (2.2.1.pre1).
[AVHWDeviceContext @ 0x1d27600] Driver not found in known nonstandard list, using standard behaviour.
[mjpeg_qsv @ 0x1d22f80] Error querying encoder params: unsupported (-3)
Error initializing output stream 0:0 -- Error while opening encoder for output stream #0:0 - maybe incorrect parameters such as bit_rate, rate, width or height
[AVIOContext @ 0x1d24680] Statistics: 0 seeks, 0 writeouts
[AVIOContext @ 0x1d11940] Statistics: 1382400 bytes read, 0 seeks
Conversion failed!
```

uartie, 2018-10-19:
Prior to the regressing commit, encode would succeed as long as ffmpeg is patched with https://patchwork.ffmpeg.org/patch/10639/

uartie, 2018-10-19:
oops, I pasted wrong ffmpeg-qsv output in description initially.  Updated.

shiqiangyu, 2018-10-21:
@uartie Thanks for reporting the issue. Need to verify whether it shared the same root cause with libva-utils conformance tests, and the fix is under reviewing. Could you share the test suits to me, so that I can verify before commit to reduce the back and forth overhead for regression reporting and fixing. Thanks.

shiqiangyu, 2018-10-22:
I got below error when running the command:

> Device creation failed: -12.
> Failed to set value 'qsv=qsv:hw' for option 'init_hw_device': Cannot allocate memory
>

shiqiangyu, 2018-10-22:
Got below error

> Device creation failed: -12.
> Failed to set value 'qsv=qsv:hw' for option 'init_hw_device': Cannot allocate memory

uartie, 2018-10-22:
Are you using upstream https://ffmpeg.org/download.html#get-sources?

uartie, 2018-10-22:
FYI, my software stack is:

```
libva (master) heads/master-0-gf0dfe5e762e7  https://github.com/01org/libva
gmmlib (master) heads/master-0-g3e602a84dccd  https://github.com/intel/gmmlib
intel-media-driver (master) heads/master-0-gf746bb2bc98b  https://github.com/intel/media-driver
msdk (master) heads/master-0-g3cf63e9144b3  https://github.com/Intel-Media-SDK/MediaSDK
ffmpeg (master) heads/master-0-g18fce29d8f1b  https://git.ffmpeg.org/ffmpeg
```

My ffmpeg configure flags are:
```
--disable-static --enable-shared --enable-libdrm --enable-vaapi --enable-libmfx \
--disable-amf --disable-audiotoolbox --disable-cuda --disable-cuda-sdk --disable-cuvid \
--disable-d3d11va --disable-dxva2 --disable-libnpp --disable-mmal --disable-nvdec \
--disable-nvenc --disable-omx --disable-omx-rpi --disable-rkmpp --disable-v4l2-m2m \
--disable-vdpau --disable-videotoolbox --enable-nonfree --enable-gpl
```

shiqiangyu, 2018-10-24:
managed to repro this issue using CNL, found the same root cause with libva-utils conformance tests, will check in soon.

uartie, 2018-10-26:
3c55dfe3061e49a3507d50cf60dafe12583b506c reverts the regressing commit. Please reference issue numbers (e.g. #373) in commit messages. GitHub will automatically create link between issue and commit and makes it easier to see when activity occurs for specific issue.

closed 2019-08-29 by XinfengZhang