intel/media-driver #375: [gst-vaapi][ICL] vaCreateImage can not used ayuv(8bit/444sample) format to create image.

feiwan1, 2018-10-23:
By using ayuv format in vaCreateImage, un-implemented status will be return. This format required by Iotg, may need your help to support this format.

feiwan1, 2018-12-11:
Reproduce step:
using master of gstramer-vaapi and cmdline:
$ gst-launch-1.0 filesrc location=main_444_8bit.h265 ! h265parse ! vaapih265dec ! fakesink

wangyan42164, 2018-12-11:
@feiwan1 The format error is gone after I applied #425. But I am not sure if it is gst-vaapi issue like the following log:
[LIBVA]:ENTER    - DdiMedia_MapBufferInternal
0:00:00.076540831  3717      0x18680f0 ERROR                  vaapi gstvaapidecoder_h265.c:2264:decode_picture: failed to reset quantizer matrix
0:00:00.076559255  3717      0x18680f0 ERROR            vaapidecode gstvaapidecode.c:755:gst_vaapidecode_handle_frame: decode error 10
ERROR: from element /GstPipeline:pipeline0/GstH265Parse:h265parse0: Internal data stream error.
Additional debug info:
gstbaseparse.c(3634): gst_base_parse_loop (): /GstPipeline:pipeline0/GstH265Parse:h265parse0:
streaming stopped, reason not-supported (-6)
ERROR: pipeline doesn't want to preroll.

wangyan42164, 2018-12-11:
@feiwan1 ensure_quant_matri() in gstvaapidecoder_h265.c will return GST_VAAPI_DECODER_STATUS_ERROR_UNSUPPORTED_CHROMA_FORMAT.
From comments, it should only support 4:2:0.

feiwan1, 2018-12-11:
Hi @wangyan42164 , you can use TSCTX_8bit_RExt_SHARP_1.bin this stream to reproduce this issue.

PR #458 (2019-04-12): Add AYUV image format.
Fixes #375.
commit 2a70dd2b (2019-04-12): Add AYUV image format.

Fixes #375.
closed 2019-04-12 by intel-mediadev