intel/media-driver #376: Does the driver support Gemini Lake?

dmitryermilov, 2018-10-23:


XinfengZhang, 2019-08-29:
not yet. the device ID should already added, but not fully test.

zcwang, 2020-04-06:
Hi All,
Does intel-media-19.4.0 (or the latest upstream of media-driver) support GLK in Linux?
https://github.com/intel/media-driver/releases/tag/intel-media-19.4.0r

Thanks!
Gary

zcwang, 2020-04-06:
[GST-error-in-GLK_gary0406.log](https://github.com/intel/media-driver/files/4439260/GST-error-in-GLK_gary0406.log)
Error caused by GST plug-in vaapi in GLK/Ubuntu 18.04

"gst-launch-1.0 -ev videotestsrc num-buffers=600 ! video/x-raw,width=3840,height=2160,framerate=30/1 ! vaapih264enc ! h264parse ! mp4mux ! filesink location=test.mp4"

zcwang, 2020-04-07:
Related issue for reference
https://github.com/intel/media-driver/issues/897

THX723, 2020-07-13:
@XinfengZhang @dmitryermilov Curious if a timeline has been established for when GLK will be fully integrated and tested in the media driver?

XinfengZhang, 2020-07-16:
#1005 should fixed some issue with GLK

XinfengZhang, 2020-10-19:
@zcwang hi Gary, have you tried PR #1005 ?

Jexu, 2022-12-02:
As mentiosned above, this issue should have been fixed. So i will close this issue and feel free to re-open it agsin if you have any other questions. Thanks

closed 2022-12-02 by Jexu