intel/media-driver #377: For decoders, vaCreateConfig() returns INVALID_PARAMETER if no attributes supplied.

djie1, 2018-10-24:
For the decoding, if vaCreateConfig() is called with (attrib_list == NULL && num_attribs == 0), it returns INVALID_PARAMETER.

uartie, 2018-10-24:
AFAIK, this is being addressed by @shiqiangyu soon... see patch in #363 and other related regressions caused by this in #372 and #373.

djie1, 2018-10-25:
@uartie , correct, patch in #363 could fix this issue and the encoding related regressions.

uartie, 2018-10-26:
3c55dfe3061e49a3507d50cf60dafe12583b506c reverts the regressing commit. Please reference issue numbers (e.g. #377) in commit messages. GitHub will automatically create link between issue and commit and makes it easier to see when activity occurs for specific issue.

closed 2019-08-29 by XinfengZhang