intel/media-driver #379: Different QVBR settings lead to the same quality

dvrogozh, 2018-10-24:
Discussed within #319.

@ChenXiaomin36 : I see some strange behaviour on enabled qvbr. Could you, please, check on your side? Consider the following mediasdk command lines with _different_ bitrate settings (compare: -b 2500 and -b 4000, -MaxKbps 3000 and MaxKbps 4800):
```
./share/mfx/samples/sample_encode h264 -i 720p.yuv -o 30_se.h264 -w 1280 -h 720 -f 30 -b 2500 -u 1 -hw -vaapi -n 600 -r 1 -g 30 -MaxKbps 3000 -qvbr 30

./share/mfx/samples/sample_encode h264 -i 720p.yuv -o 48_se.h264 -w 1280 -h 720 -f 30 -b 4000 -u 1 -hw -vaapi -n 600 -r 1 -g 30 -MaxKbps 4800 -qvbr 30
```

The strange thing is that if you will check the quality of output streams, you will see that PSNR values are identical. Are there any limit in the driver which forces bitrates to be equal to this limit if user-specified bitrate is lower than this limit? I am looking into such a limit in mediasdk, but it seems this can be something on the driver side, because settings which mediasdk pass to VAEncMiscParameterRateControl look correct (my custom debug prints below added here: https://github.com/Intel-Media-SDK/MediaSDK/blob/master/_studio/mfx_lib/shared/src/mfx_h264_encode_vaapi.cpp#L210):
```
>>> rate_param->bits_per_second=2999296, rate_param->window_size=300000, rate_param->target_percentage=83, rate_param->quality_factor=30

>>> rate_param->bits_per_second=4799488, rate_param->window_size=480000, rate_param->target_percentage=83, rate_param->quality_factor=30
```

Note: you need this PR Intel-Media-SDK/MediaSDK#656 to try QVBR under mediasdk.

Xiaogangli-intel, 2020-06-22:
@dvrogozh so, this issue should be closed by  #319, right?

Xiaogangli-intel, 2022-04-01:
As @kovakimy said, this issue is not reproduced with latest env. I close this issue, if you still have other concern, let me know please.