intel/media-driver #389: AVC LP Multi-Slice Encode corrupted

uartie, 2018-10-29:
Encoding h264 low-power with multi-slice produces corrupted output.

**Reproduce Steps:**

In gst-msdk:
```
GST_VAAPI_ALL_DRIVERS=1 LIBVA_DRIVER_NAME=iHD gst-launch-1.0 videotestsrc num-buffers=100 \
  ! video/x-raw,format=NV12 ! msdkh264enc rate-control=cqp qpp=28 target-usage=4 \
  gop-size=30 num-slices=4 b-frames=0 hardware=true low-power=1 \
  ! video/x-h264,profile=high ! h264parse ! filesink location=test.h264
```

In gst-vaapi:
```
GST_VAAPI_ALL_DRIVERS=1 LIBVA_DRIVER_NAME=iHD gst-launch-1.0 videotestsrc num-buffers=100 \
  ! video/x-raw,format=NV12 ! vaapih264enc rate-control=cqp init-qp=28 quality-level=4 \
  keyframe-period=30 num-slices=4 max-bframes=0 tune=3 \
  ! video/x-h264,profile=high ! h264parse ! filesink location=test.h264
```

In ffmpeg-vaapi:
```
LIBVA_DRIVER_NAME=iHD ffmpeg -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -v verbose \
  -f rawvideo -pix_fmt yuv420p -s:v 1280x720 -i input.yuv -vf 'format=nv12,hwupload' \
  -c:v h264_vaapi -low_power 1 -profile:v high -g 30 -qp 28 -bf 0 -quality 4 -slices 4 \
  -vframes 100 -y test.h264
```

Playback test.h264:
```
LIBVA_DRIVER_NAME=iHD GST_VAAPI_ALL_DRIVERS=1 gst-launch-1.0 filesrc location=test.h264 \
  ! h264parse ! vaapih264dec ! vaapisink
```
See more details in https://trac.ffmpeg.org/ticket/7523

Software Stack:
```
libva (master) heads/master-0-g285267586a3d  https://github.com/01org/libva
gmmlib (master) heads/master-0-gaedd82e2da01  https://github.com/intel/gmmlib
intel-media-driver (master) heads/master-0-g9d258e13eba1  https://github.com/intel/media-driver
gstreamer (master) heads/master-0-g3eb334a7f9a5  https://anongit.freedesktop.org/git/gstreamer/gstreamer
gst-plugins-base (master) heads/master-0-ga15baf797603  https://anongit.freedesktop.org/git/gstreamer/gst-plugins-base
gst-plugins-good (master) heads/master-0-g560a6155bd5b  https://anongit.freedesktop.org/git/gstreamer/gst-plugins-good
gst-plugins-bad (master) heads/master-0-gafbbc3a97ec3  https://anongit.freedesktop.org/git/gstreamer/gst-plugins-bad
gstreamer-vaapi (master) heads/master-0-gd8442e479ea8  https://anongit.freedesktop.org/git/gstreamer/gstreamer-vaapi
ffmpeg (master) heads/master-0-g4a6c286acc45  https://git.ffmpeg.org/ffmpeg
```

wangyan42164, 2018-10-31:
@uartie Could you please clarify it. GST is OK and FFMPEG fails? Or both fails. And what platform do you use? KBL or SKL?
Thanks

uartie, 2018-10-31:
It fails with both FFMPEG and GST (i.e. all commands above under "Reproduce Steps").  I am using KBL locally, but our CI shows same issue exists on KBL, CFL and WHL.

uartie, 2018-10-31:
I have not tested it on SKL since the intel-media-driver docs don't include AVC E2 (LP) support.

wangyan42164, 2018-10-31:
@uartie Thanks for your clarifying. I will check it.

mypopydev, 2018-11-03:
@wangyan42164 @uartie  SKL failed too

XinfengZhang, 2018-11-12:
@wangyan42164 please help to check the slice structure , for Low power mode, iHD just support equal_rows slice structure.

wangyan42164, 2018-11-15:
@XinfengZhang @uartie @mypopydev 
In fact, equal_row slice may be too strict. It should be enough for row aligned. And current media driver will force one slice for one row if gstreamer/ffmpeg send non row-aligned slice parameter. 
The root cause is m_isVdencSuperSliceEnabled is enabled in media driver when using low power encoding. For super slice, first_mb_in_slice of slice header should be zero always. 
I have submitted one patch for gst-vaapi: https://gitlab.freedesktop.org/gstreamer/gstreamer-vaapi/merge_requests/15
ffmpeg should be similar. I will also submit one for this.
msdk may also need one? I will check it.
Thanks.

xhaihao, 2018-11-15:
How is  the middle-ware aware of the driver's limitation? Your patch (https://gitlab.freedesktop.org/gstreamer/gstreamer-vaapi/merge_requests/15) doesn't make sense to me

xhaihao, 2018-11-15:
it makes sense first_mb_in_slice is 0 for the first slice, but for other slices, first_mb_in_slice shouldn't be 0.

wangyan42164, 2018-11-15:
@xhaihao @XinfengZhang Could we add one query API into libva for super slice usage? We may need more discussion.

uartie, 2019-04-30:
This seems to be fixed in _gst-msdk_ and _ffmpeg-qsv_, but _gst-vaapi_ and _ffmpeg-vaapi_ still has this issue.  I'm not sure when or what fixed this in gst-msdk/ffmpeg-qsv.  My current stack is:
```
libva (master) heads/master-0-g4d9d819d4859 
gmmlib (master) heads/master-0-g0e5e49651e7d 
intel-media-driver (master) heads/master-0-g8018ad53fc9e 
msdk (master) heads/master-0-g2f0a8af81330 
ffmpeg (master) heads/master-0-gd0e4d0429ed2
gstreamer (master) heads/master-0-gab60a330db71 
gst-plugins-base (master) heads/master-0-g123c2f8b5dd4 
gst-plugins-good (master) heads/master-0-g037d70c01bbe 
gst-plugins-bad (master) heads/master-0-g3a0a82a783b2 
gstreamer-vaapi (master) heads/master-0-ga3ce4446c168 
```
Can someone confirm and/or point to commit that fixed it in above 2 cases?

uartie, 2019-04-30:
@dvrogozh are you aware of any specific fix/commit in the msdk component that would have fixed this in gst-msdk and ffmpeg-qsv (see my last comment).

XinfengZhang, 2019-07-29:
how about driver handle it, such as decode the slice parameter from the packed header then pack it again inside driver？

wangyan-intel, 2019-07-29:
@XinfengZhang Yes. I also think so. I am working on this.

wangyan42164, 2019-07-30:
@uartie I have submitted #697 to fix it on media driver side. Now ffmpeg/gst-vaapi can work fine for the case. BTW, I find that MSDK also set first_mb_in_slice to 0 in his implementation. So ffmpeg/gst use MSDK has not this issue.

Now I tried your decoding command to decode the output H264 stream. It displays rightly.
But I still get failure on my local KBL platform by your https://github.com/intel/vaapi-fits like the following:
./vaapi-fits run test/gst-vaapi/encode/avc.py:cqp:lp.test -k 'slices=4'
It shows PSNR difference:
148727 [2019-07-30 07:34:38.438865] NOTICE: slash: DETAIL: time(gst:2) = 11.8865s
148728 [2019-07-30 07:34:38.506857] NOTICE: slash: DETAIL: time(psnr:1) = 0.0678s
148729 [2019-07-30 07:34:38.507039] NOTICE: slash: DETAIL: drv.iHD.psnr:expect = None
148730 [2019-07-30 07:34:38.507113] NOTICE: slash: DETAIL: drv.iHD.psnr:actual = [40.9678, 60.4905, 58.777, 56.6767, 61.0828, 59.622]
148731 [2019-07-30 07:34:38.528668] TRACE: slash.core.error: Error added: AssertionError: Invalid reference value
I am not sure the path of generating the golden data? We may need compare it and AVC VDENC.

Currently AVC VDENC will adjust slice layout dynamically if ffmpeg/gst-vaapi set multi-slice but cannot meted. 
E.g. When set slices = 4, resolution=320x240, AVC VDENC will changes to row slice layout automatically.
If need 4 slices, you have to set the resolution to 320x256. MSDK will do this padding.
So I think ffmpeg/gst-vaapi need some additional logic. I will try to fix it in ffmpeg/gst-vaapi.
Thanks

Yan Wang

uartie, 2019-08-16:
@wangyan42164 I can update vaapi-fits once your patch is merged.

zhuqingliang, 2019-09-05:
some avc cbr vdenc still failed on my ICL:
  Cmd: ffmpeg -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -v verbose -f rawvideo -pix_fmt yuv420p -s:v 720x480 -r:v 30 -i  ./yuv/720x480p.yuv -vf 'format=nv12,hwupload' -c:v h264_vaapi -profile:v high -rc_mode CBR -g 60 -slices 3 -b:v 2000k -maxrate 2000k -low_power 1 -vframes 150 -y ./480p-cbr-high-30-60-3-2000k-2000k-1.h264

message:
[h264_vaapi @ 0x118ab40] Failed to end picture encode issue: 24 (internal encoding error).
[h264_vaapi @ 0x118ab40] Encode failed: -5.
Video encoding failed

PR #697 (2019-10-12): Adjust packed slice header for AVC VDENC when multi-slice encoding.
AVC VDENC when multi-slice encoding need force first_mb_in_slice to 0.
Fixes #389.
commit 8c83e39f (2019-10-12): Adjust packed slice header for AVC VDENC when multi-slice encoding.

AVC VDENC when multi-slice encoding need force first_mb_in_slice to 0.
Fixes #389.
PR #785 (2019-11-26): Adjust packed slice header for AVC VDENC when multi-slice encoding.
AVC VDENC when multi-slice encoding need force first_mb_in_slice to 0.
Fixes #389.
commit 4073674d (2019-11-26): Adjust packed slice header for AVC VDENC when multi-slice encoding.

AVC VDENC when multi-slice encoding need force first_mb_in_slice to 0.
Fixes #389.
closed 2019-10-12 by intel-mediadev