intel/media-driver #393: Render target table implementation is faulty

vshampor, 2018-10-31:
Currently, when the render target table is full and a new surface needs to be inserted into it via DdiMediaBase::RegisterRTSurfaces, the table will search for the first inactive surface from the beginning of the table's static array and remove it to insert the new one. This is wrong, since the first inactive surface from the beginning of the table's static array is not necessarily the oldest and may, in fact, not be active in the current or previous frame and be used as a reference for the next frames. 

In case the table is full, there is no way around evicting a surface - the table should make sure that this is the oldest registered surface.

vshampor, 2018-10-31:
In fact we already have tests failing on MediaSDK side due to this issue - when we allocate more HW surfaces on app level via VA calls than `DDI_MEDIA_MAX_SURFACE_NUMBER_CONTEXT` (which is 127) and run decode, we get a MFX_ERR_DEVICE_FAILED at Mos_Specific_GetResourceInfo.

I provided a solution that fixes the issue in the following PR:
https://github.com/intel/media-driver/pull/392

A more elegant solution would be to use some sort of an STL container. However, judging from the overall C-style appearance of the code, I was reluctant to use STL there and instead went for the C-style approach. Please advise if the PR should be redesigned to use the C++ approach instead. Any feedback for the PR would also be welcome.

XinfengZhang, 2018-11-04:
thanks @vshampor, will check your PR and test it.

commit 2ab0b534 (2018-12-22): [VP9d] Implement continuous ring buffer logic for render target table

Fixes: #393
closed 2018-12-22 by Sherry-Lin