intel/media-driver #405: Media driver compilation fails with- "fatal error: va/va_dricommon.h: No such file or directory"

ajitkumarkp, 2018-11-05:
Hi, 
I have successfully installed gmmlib and libva. But when it comes to the media-driver, the installation fails with the below error messages. Please help to resolve. Let me know if/what additional info is needed from my side. Tx.

[ 30%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/linux/common/os/libdrm/xf86drmHash.c.o
[ 30%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/linux/common/os/libdrm/xf86drmMode.c.o
[ 31%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/linux/common/os/libdrm/mos_bufmgr.c.o
[ 31%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/linux/common/os/libdrm/mos_bufmgr_api.c.o
In file included from /home/intelpae/Desktop/MediaStack/media-driver/media_driver/linux/common/ddi/media_libva_putsurface_linux.cpp:46:0:
**/home/intelpae/Desktop/MediaStack/media-driver/media_driver/linux/common/ddi/media_libva_putsurface_linux.h:31:10: fatal error: va/va_dricommon.h: No such file or directory
 #include <va/va_dricommon.h>
          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/build.make:10526: recipe for target 'media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/linux/common/ddi/media_libva_putsurface_linux.cpp.o' failed
make[2]: *** [media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/linux/common/ddi/media_libva_putsurface_linux.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /home/intelpae/Desktop/MediaStack/media-driver/media_driver/linux/common/ddi/media_libva.cpp:45:0:
/home/intelpae/Desktop/MediaStack/media-driver/media_driver/linux/common/ddi/media_libva_putsurface_linux.h:31:10: fatal error: va/va_dricommon.h: No such file or directory
 #include <va/va_dricommon.h>
          ^~~~~~~~~~~~~~~~~~~
compilation terminated.**
media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/build.make:10430: recipe for target 'media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/linux/common/ddi/media_libva.cpp.o' failed
make[2]: *** [media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/linux/common/ddi/media_libva.cpp.o] Error 1
CMakeFiles/Makefile2:167: recipe for target 'media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/all' failed
make[1]: *** [media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2

XinfengZhang, 2018-11-08:
hi , could you help to check is va_dricommon.h is exist in your installed libva include folder?

ajitkumarkp, 2018-11-08:
Hi, Yes va_dricommon.h exists in <workspace>/libva/va folder.
Here's my tree:
<workspace>
├── build_media
├── gmmlib
├── libva
├── libva-utils
└── media-driver

eero-t, 2018-11-16:
Do you have libva-dev package installed on your distribution?  If yes, check that media driver didn't find that instead of the one you built from your CMake configuration (if that's the issue, you didn't tell media driver where to find the installed config files for the sources you built).

ajitkumarkp, 2018-11-21:
Sry for the delayed response! I checked, I dont have libva-dev pkg. btw im using Ubuntu 17.10.

ajitkumarkp, 2018-11-21:
The problem was with my libva installation path. Did the whole thing from scratch and its working this time. Closing the issue, Thanks!

closed 2018-11-21 by ajitkumarkp

okias, 2020-04-09:
Got the same error, when libva is compiled _without_ X support (running stack on wayland). Is this header also required for wayland build?

okias, 2020-09-19:
Issue is still valid on 20.2.0 .

XinfengZhang, 2020-09-21:
same as https://github.com/intel/media-driver/issues/308#issuecomment-529856356