intel/media-driver #409: [ICL][ffmpeg-vaapi][VP8 encode] ffmpeg exits with "Output file is empty, nothing was encoded"

ykku16github, 2018-11-08:
[201811081026_vp8_320x180_100kbps_30fps.txt](https://github.com/intel/media-driver/files/2563135/201811081026_vp8_320x180_100kbps_30fps.txt)

The attached stdout was captured by running the following ffmpeg command:
`ffmpeg -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -f rawvideo -pix_fmt nv12 -s:v 320x180 -r 30 -i tulip2-320x180-55be7124b3aec1b72bfb57f433297193.yuv -vf 'format=nv12,hwupload' -c:v vp8_vaapi -b:v 100k -maxrate 100k -y -r 30 -vstats -v debug output.webm 2>&1 | tee 201811081026_vp8_320x180_100kbps_30fps.txt`

If needed, the input file can be downloaded from:
[http://commondatastorage.googleapis.com/chromiumos-test-assets-public/tulip2/tulip2-320x180-55be7124b3aec1b72bfb57f433297193.yuv](url)

ykku16github, 2018-11-08:
For every frame that gets encoded, ffmpeg prints the following:
`[vp8_vaapi @ 0x27dee40] Output buffer: 32717 bytes (status 00000001).
[vp8_vaapi @ 0x27dee40] Output buffer: 0 bytes (status 00000000).
`
This is printed by the for-loop in:
[https://github.com/FFmpeg/FFmpeg/blob/70463b702697ef8be4ec4d88969b4fce5b4e7a7e/libavcodec/vaapi_encode.c#L517](url)

This is because vaMapBuffer() returned 2 packets, caused by the following code in iHD:
[https://github.com/intel/media-driver/blob/f3824c8ed4ee02ecb6481dbd7d21225f26454307/media_driver/linux/common/codec/ddi/media_ddi_encode_vp8.cpp#L244](url)

It's not clear to me whether the proper fix should be in ffmpeg or in iHD.  Any thoughts?

Tianhaol, 2018-11-22:
This issue can be reproduced on our side.

eero-t, 2019-05-14:
Issue title is misleading.  This issue isn't ICL specific, it happens also e.g. on KBL.

zhoub, 2019-05-20:
I have this fixed in FFmpeg as workaround.

https://github.com/zhoub/FFmpeg/commit/493e507e20663aec29cfb699e82a98dad70ed088

wangyan-intel, 2019-08-29:
This is ffmpeg issue.

closed 2019-08-29 by wangyan-intel

eero-t, 2019-08-29:
> This is ffmpeg issue.

There was a comment that it's an issue between media-driver and FFmpeg, and it's not clear which should be fixed.

You would need to make sure there's a bug reported for FFmpeg (link please!) and it's been accepted as FFmpeg issue, before closing this.

wangyan-intel, 2019-08-30:
Media driver will return 2 buffers. One is coded data buffer, another is some additional suggestion info for future encoding. So ffmpeg should ignore the 2nd buffer if it doesn't care theses suggested info.