intel/media-driver #410: [ICL][ffmpeg-vaapi][VP8 encode] Encode bitrates are much greater than requested max bitrates

ykku16github, 2018-11-08:
[vstats_210542.log](https://github.com/intel/media-driver/files/2563842/vstats_210542.log)
[201811072105_vp8_320x180_100kbps_30fps.txt](https://github.com/intel/media-driver/files/2563843/201811072105_vp8_320x180_100kbps_30fps.txt)

The attached stdout and vstats log were captured by running the following command:
`ffmpeg -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -f rawvideo -pix_fmt nv12 -s:v 320x180 -r 30 -i tulip2-320x180-55be7124b3aec1b72bfb57f433297193.yuv -vf 'format=nv12,hwupload' -c:v vp8_vaapi -b:v 100k -maxrate 100k -y -r 30 -vstats output.webm 2>&1 | tee 201811072105_vp8_320x180_100kbps_30fps.txt`

The input file can be downloaded from:
[http://commondatastorage.googleapis.com/chromiumos-test-assets-public/tulip2/tulip2-320x180-55be7124b3aec1b72bfb57f433297193.yuv](http://commondatastorage.googleapis.com/chromiumos-test-assets-public/tulip2/tulip2-320x180-55be7124b3aec1b72bfb57f433297193.yuv)

Note that the requested max bitrate is 100k.  But the last line of the attached vstats log shows that the achieved average bitrate is 954.2k.

XinfengZhang, 2018-11-12:
thanks , will check it. thanks

XinfengZhang, 2018-11-22:
@Tianhaol , please update it.

Tianhaol, 2018-11-27:
@ykku16github, could you provide the patch to dump vp8 file?  Because issue 409 dumped the empty file.

ykku16github, 2018-11-27:
I worked around issue 409 by doing the following:
`diff --git a/media_driver/linux/common/codec/ddi/media_ddi_encode_vp8.cpp b/media_driver/linux/common/codec/ddi/media_ddi_encode_vp8.cpp
index f2a2202..5be510e 100644
--- a/media_driver/linux/common/codec/ddi/media_ddi_encode_vp8.cpp
+++ b/media_driver/linux/common/codec/ddi/media_ddi_encode_vp8.cpp
@@ -240,8 +240,10 @@ VAStatus DdiEncodeVp8::StatusReport(
             // if HDCP2 is enabled, the second segment for counter values is already there  So, move the connection as the third one
             if (m_encodeCtx->pCpDdiInterface->IsHdcp2Enabled())
                 pFindTheLastEntry = (VACodedBufferSegment *)pFindTheLastEntry->next;
+#if 0
             // connect Status report here
             pFindTheLastEntry->next = pVACodedBufferSegmentForStatusReport;
+#endif
 
             break;
         }
`

Tianhaol, 2018-11-28:
Thanks. An explanation are required to provide why  pFindTheLastEntry->next  is not null. We're figuring out how to handle it.

Tianhaol, 2018-11-28:
AP will read the next segment for long-term reference enabling control. Otherwise, please discard the next segment.

Tianhaol, 2018-11-29:
Please help to verify with this PR https://github.com/intel/media-driver/pull/433

ykku16github, 2018-11-29:
Thanks.  Your patch works.  The same ffmpeg command (from my first comment) now shows an average bitrate of 102.7k.

eero-t, 2019-03-01:
> Thanks. Your patch works. The same ffmpeg command (from my first comment) now shows an average bitrate of 102.7k.

If the issue is fixed, could somebody close this issue?

closed 2019-08-29 by XinfengZhang