intel/media-driver #411: MFE does not support some features

dvrogozh, 2018-11-09:
See https://github.com/Intel-Media-SDK/MediaSDK/issues/863 - GPU happens on ffmpeg-qsv if MFE is used for ICQ path. There are other cases which are not supported by MFE:
* ICQ
* QVBR
* VCM

and some other, they are highlighted inhttps://github.com/Intel-Media-SDK/MediaSDK/pull/882.

As of now VAAPI does not have any query API to return list of features supported or not by MFE, for example vaQueryConfigAttributes know nothing about MFE. In a way this means that everything supported by legacy encoder _should_ be supported by MFE and actually this makes sense.

Could you, please, fix MFE behavior and support currently not supported features? As I noted expectation is that _everything_ supported by legacy encoder is accessible via MFE. This should b take care of.

dvrogozh, 2018-11-09:
@lizhong1008, @artem-shaporenko  : FYI

artem-shaporenko, 2018-11-12:
> Yes, we probably need some way to to query capabilities assuming we will have MFE. Second thing is that I would like to see ICQ and QVBR and other features be supported by MFE if that's theoretically possible.

ICQ is definitely possible and requires support. For QVBR is it supported by Lowpower encoder only or by dual pipe as well?

artem-shaporenko, 2018-11-12:
Right now I'm thinking that there is just a bug for MFE during BRC kernel  integration with support for ICQ, rather then issue with supporting MFE, should be easy to fix.

dvrogozh, 2018-11-12:
> For QVBR is it supported by Lowpower encoder only or by dual pipe as well?
QVBR is supported in DP encoder (see _merged_ https://github.com/intel/media-driver/pull/319). And as of now it is not supported in LP encoder, but patch seems to be on its way, see https://github.com/intel/media-driver/pull/404 (not merged).

artem-shaporenko, 2018-11-12:
Ok, in this case QVBR should be also supported.

artem-shaporenko, 2018-11-12:
BTW anyone checked LP encoder is not trying to work in MFE mode?

dvrogozh, 2018-11-12:
I did not. And I believe that in mediasdk we just enabled DP QVBR support, LP is missing.

artem-shaporenko, 2018-11-12:
Actually usage wise - LP is more priority for QVBR usage.

XinfengZhang, 2020-07-04:
@artem-shaporenko , how about current status?  suppose legacy feature should also be supported by MFE, is it a bug?

XinfengZhang, 2023-01-28:
I will close it as it is aged issue and lost track. 
a quick summary:
MFE was expected to support all legacy feature , because it does not have independent capability query interfaces , it reuse legacy one. 
so , we have 3 choices:
1. add other interfaces for MFE in libva for the caps query
2. add full support for MFE
3. defer it , treat it as a special feature to some customization needed. so just need call out the limitation in document. 
I prefer 3rd one. so, I will close it , and will add it into known issue list.

closed 2023-01-28 by XinfengZhang