intel/media-driver #415: [vo/gpu/vaapi-egl] vaAcquireSurfaceHandle() failed (invalid VASurfaceID)

enihcam, 2018-11-13:
Intel(R) Core(TM) i5-7200U CPU @ 2.50GHz
Archlinux (with all packages up to latest)
intel-media-driver ver: 18.3.0

mpv output:
```
[vo/gpu/vaapi-egl] vaAcquireSurfaceHandle() failed (invalid VASurfaceID)
[vo/gpu/vaapi-egl] vaAcquireSurfaceHandle() failed (invalid VASurfaceID)
[vo/gpu/vaapi-egl] vaAcquireSurfaceHandle() failed (invalid VASurfaceID)
[vo/gpu/vaapi-egl] vaAcquireSurfaceHandle() failed (invalid VASurfaceID)
[vo/gpu/vaapi-egl] vaAcquireSurfaceHandle() failed (invalid VASurfaceID)
[vo/gpu/vaapi-egl] vaAcquireSurfaceHandle() failed (invalid VASurfaceID)
[vo/gpu/vaapi-egl] vaAcquireSurfaceHandle() failed (invalid VASurfaceID)
[vo/gpu/vaapi-egl] vaAcquireSurfaceHandle() failed (invalid VASurfaceID)
[vo/gpu/vaapi-egl] vaAcquireSurfaceHandle() failed (invalid VASurfaceID)
[vo/gpu/vaapi-egl] vaAcquireSurfaceHandle() failed (invalid VASurfaceID)
Using hardware decoding (vaapi).
VO: [gpu] 1920x1080 vaapi[nv12]
AO: [pulse] 44100Hz stereo 2ch float
AV: 00:00:15 / 00:14:06 (1%) A-V:  0.000
```

XinfengZhang, 2018-11-22:
@enihcam it seems a old question, no vaAcquaieSurfaceHandle interface in libva 2.x
suggest to use vaExportSurfaceHandle

enihcam, 2018-11-22:
Then when (which future version) will vaAcquaieSurfaceHandle be available?

XinfengZhang, 2019-08-29:
close it, for there are no such api inside libva https://github.com/intel/libva

closed 2019-08-29 by XinfengZhang