intel/media-driver #432: [ffmpeg-vaapi][hevc 10bit encode] HEVC 10bit encode CMD line failed on iHD, but work for i965

FocusLuo, 2018-11-27:
These CMD can't work on ffmpeg-vaapi+i965, but failed on ffmpeg-vaapi+iHD
CQP:
ffmpeg -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -v verbose -f rawvideo -s:v 1920x1080 -i /media/encoderbitstreams/10bit/1080P_1920x1080.P010 -vf format=p010,hwupload -c:v hevc_vaapi  -profile:v main10 -g 30 -bf 3 -qp 28 -quality 4 -vframes 100 -y ./hevce_cqp_1920x1080_gop30_refdist3_qp28.h265
error info:
Press [q] to stop, [?] for help
[graph 0 input from stream 0:0 @ 0x17ab3c0] w:1920 h:1080 pixfmt:yuv420p tb:1/25 fr:25/1 sar:0/1 sws_param:flags=2
[auto_scaler_0 @ 0x17ac880] w:iw h:ih flags:'bicubic' interl:0
[Parsed_format_0 @ 0x17aa180] auto-inserting filter 'auto_scaler_0' between the filter 'graph 0 input from stream 0:0' and the filter 'Parsed_format_0'
[auto_scaler_0 @ 0x17ac880] w:1920 h:1080 fmt:yuv420p sar:0/1 -> w:1920 h:1080 fmt:p010le sar:0/1 flags:0x4
[hevc_vaapi @ 0x17a6900] Input surface format is p010le.
[hevc_vaapi @ 0x17a6900] Using VAAPI profile VAProfileHEVCMain10 (18).
[hevc_vaapi @ 0x17a6900] Using VAAPI entrypoint VAEntrypointEncSlice (6).
[hevc_vaapi @ 0x17a6900] Using VAAPI render target format YUV420_10 (0x100).
[hevc_vaapi @ 0x17a6900] RC mode: VBR, 50% of 10000000 bps over 500 ms.
[hevc_vaapi @ 0x17a6900] RC buffer: 5000000 bits, initial fullness 3750000 bits.
[hevc_vaapi @ 0x17a6900] Using intra, P- and B-frames (supported references: 3 / 1).
[hevc_vaapi @ 0x17a6900] All wanted packed headers available (wanted 0xd, found 0x1f).
[hevc_vaapi @ 0x17a6900] Using level 4.
Output #0, hevc, to './hevce_vbr_1920x1080_gop30_refdist3_bitrate5M.h265':
  Metadata:
    encoder         : Lavf58.20.100
    Stream #0:0: Video: hevc (hevc_vaapi) (Main 10), 1 reference frame, vaapi_vld, 1920x1080, q=-1--1, 5000 kb/s, 25 fps, 25 tbn, 25 tbc
    Metadata:
      encoder         : Lavc58.35.100 hevc_vaapi
[hevc_vaapi @ 0x17a6900] Failed to end picture encode issue: 24 (internal encoding error).
[hevc_vaapi @ 0x17a6900] Encode failed: -5.
Video encoding failed
[AVIOContext @ 0x17a7400] Statistics: 0 seeks, 1 writeouts
[AVIOContext @ 0x17a3a40] Statistics: 15552000 bytes read, 0 seeks
Conversion failed!

The CBR/VBR mode, also have this issue
CBR:
ffmpeg -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -v verbose -f rawvideo -s:v 1920x1080 -i /media/encoderbitstreams/10bit/1080P_1920x1080.P010 -vf format=p010le,hwupload -c:v hevc_vaapi -profile:v main10 -g 30 -bf 3 -b:v 5M -maxrate:v 5M -quality 4 -vframes 200 -y ./hevce_cbr_1920x1080_gop30_refdist3_bitrate5M.h265
VBR:
ffmpeg -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -v verbose -f rawvideo -s:v 1920x1080 -i /media/encoderbitstreams/10bit/1080P_1920x1080.P010 -vf format=p010le,hwupload -c:v hevc_vaapi -profile:v main10 -g 30 -bf 3 -b:v 5M -quality 4 -vframes 100 -y ./hevce_vbr_1920x1080_gop30_refdist3_bitrate5M.h265

FocusLuo, 2018-11-27:
ffmpeg-qsv can work well on iHD for cqp/cbr/vbr hevc 10bit encode

wangyan42164, 2018-11-27:
@FocusLuo Please try #371. It has passed the review but hasn't been merged. Thanks.

wangyan-intel, 2019-06-04:
#371 has been merged. So close this issue.

closed 2019-06-04 by wangyan-intel