intel/media-driver #434: Video encoded with libx265 hangs GPU at decoding.

lvella, 2018-11-29:
Entire X11 session freezes while decoding process is running. Tested decoding with VLC, MPV and ffmpeg on processor Intel(R) Atom(TM) x5-Z8350  CPU @ 1.44GHz. Also tested on Skylake Core i3 and Core i7.

[This is a sample file](https://github.com/intel/media-driver/files/2629547/arquivo.ts.gz) encoded with libx265 on ffmpeg that triggers the problem.

XinfengZhang, 2018-12-07:
thanks for reporting the issue. we will check it.

eero-t, 2019-03-01:
FFmpeg shows errors for the attached HEVC video both with CPU decoder and iHD VAAPI driver, so it appears at least partly broken.  There are no GPU hangs.

If there are GPU hangs with x5-Z8350, which is BSW / Cherry trail processor which isn't supported by this driver (media-driver i.e. iHD), those hangs came from the old i965 driver.  And indeed, using i965 driver in Ubuntu 18.04 causes a GPU hang.  i965 issue tracker has e.g. this bug for HEVC decoding:
https://github.com/intel/intel-vaapi-driver/issues/264

=> INVALID.  Please close this bug, and add your issue to the correct issue tracker.

closed 2019-08-29 by wangyan-intel