intel/media-driver #436: Performance issue when dump frame to CPU with VPP/scaling.

mypopydev, 2018-12-02:
Used the command in  SKL like:

```
LIBVA_DRIVER_NAME=iHD ./ffmpeg -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -hwaccel_output_format vaapi -i /home/barry/Videos/skyfall2-trailer.mp4 \
-an -vf "scale_vaapi=1280:720:format=nv12,hwdownload" -pix_fmt nv12 -f null /dev/null
```

and 
```
LIBVA_DRIVER_NAME=i965 ./ffmpeg -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -hwaccel_output_format vaapi -i /home/barry/Videos/skyfall2-trailer.mp4 \
-an -vf "scale_vaapi=1280:720:format=nv12,hwdownload" -pix_fmt nv12 -f null /dev/null
```

i965 performance better than iHD about 5 times in mine test env for this case, I think iHD must be have some performance issue in this case.

XinfengZhang, 2018-12-07:
thanks reporting, could you help to try #408 ?

eero-t, 2019-01-29:
> i965 performance better than iHD about 5 times in mine test env for this case, I think iHD must be have some performance issue in this case.

This could be issue of kernel not keeping GPU (and CPU) speed up.  Which distro/kernel version do you use?

Do you use "powersave" P-state governor?
  `head /sys/devices/system/cpu/cpufreq/policy*/scaling_governor`

If yes, does changing to "performance" P-state governor at least double iHD performance?
```
for i in /sys/devices/system/cpu/cpufreq/policy*/scaling_governor; do
  echo performance | sudo tee $i;
done
```

XinfengZhang, 2020-03-09:
@mypopydev , the issue should already been fixed by sw swizzle

closed 2020-05-14 by XinfengZhang