intel/media-driver #439: [gstreamer-vaapi][avc encode]the bitrate of avc vbr encoded file is out of range target 10%.

wangzj0601, 2018-12-04:
System Environment
=======
Platform: KBL/APL/SKL
Arch: x86_64
Kernel:	4.18.0+
libva	commit 285267586a3d4db0e721d30d4a5f5f9fe6a3c913
media_driver	commit 59763808054c2f9f708e1c178180432d62b72c40
gstreamer master branch c45b386b54aea4e707a8be5cbf79dc43623c4614
repo: https://gitlab.freedesktop.org/gstreamer/gstreamer.git
gst-vaapi master branch c728fb4ff6c3b4fc9eb9b43fcea4a22bc15fd678
repo: https://gitlab.freedesktop.org/gstreamer/gstreamer-vaapi.git

Bug detailed description
=======
the bitrate of encoded file is out of range target 10% than predefined setting.

Reproduce Steps
==============
1. build enc as above lists
2. gst-launch-1.0 filesrc location=/media/encoderbitstreams//SOCCER_704x576_30_orig_02.yuv '!' videoparse format=i420 width=704 height=576 framerate=30 '!' videoconvert '!' video/x-raw,format=NV12 '!' vaapih264enc rate-control=vbr bitrate=1500 keyframe-period=30 num-slices=1 max-bframes=0 quality-level=1 '!' video/x-h264,profile=main '!' h264parse '!' filesink location=./SOCCER_704x576_i420_profilemain_rc1_ip_bps1500_slice1_gop30_tu1_fps30.h264
3. bitrate_real=1067.33
4. bitrate_gap=28.00

Media_AVC_Enc_VBR/SOCCER_704x576_i420_profilemain_rc1_ip_bps1500_slice1_gop30_tu1_fps30.h264 VBR 1500 1067.33 28.00 FAIL
Media_AVC_Enc_VBR/SOCCER_704x576_i420_profilemain_rc1_ip_bps2000_slice4_gop30_tu4_fps25.h264 VBR 2000 1427.08 28.00 FAIL
Media_AVC_Enc_VBR/SOCCER_704x576_i420_profilemain_rc1_ipb_bps4000_slice7_gop30_tu7_fps30.h264 VBR 4000 2845.57 28.00 FAIL
Media_AVC_Enc_VBR/SOCCER_704x576_i420_profilehigh_rc1_ip_bps3000_slice1_gop30_tu1_fps30.h264 VBR 3000 2147.27 28.00 FAIL
Media_AVC_Enc_VBR/SOCCER_704x576_i420_profilehigh_rc1_ip_bps1500_slice4_gop30_tu4_fps25.h264 VBR 1500 1060.53 29.00 FAIL
Media_AVC_Enc_VBR/SOCCER_704x576_i420_profilehigh_rc1_ipb_bps4000_slice7_gop30_tu7_fps30.h264 VBR 4000 2845.58 28.00 FAIL
Media_AVC_Enc_VBR/SOCCER_704x576_i420_profilebaseline_rc1_ip_bps2000_slice4_gop30_tu4_fps25.h264 VBR 2000 1427.08 28.00 FAIL
Media_AVC_Enc_VBR/SOCCER_704x576_i420_profilebaseline_rc1_ip_bps3000_slice4_gop30_tu1_fps30.h264 VBR 3000 2153.02 28.00 FAIL
Media_AVC_Enc_VBR/SOCCER_704x576_i420_profilebaseline_rc1_ip_bps4000_slice4_gop30_tu7_fps30.h264 VBR 4000 2878.07 28.00 FAIL
Media_AVC_Enc_VBR/720p5994_parkrun_i420_profilemain_rc1_ip_bps1500_slice1_gop30_tu1_fps25.h264 VBR 1500 1026.02 31.00 FAIL
Media_AVC_Enc_VBR/720p5994_parkrun_i420_profilemain_rc1_ip_bps3000_slice4_gop60_tu4_fps30.h264 VBR 3000 2089.61 30.00 FAIL
Media_AVC_Enc_VBR/720p5994_parkrun_i420_profilemain_rc1_ipb_bps4000_slice8_gop59_tu7_fps30.h264 VBR 4000 2749.17 31.00 FAIL
Media_AVC_Enc_VBR/720p5994_parkrun_i420_profilehigh_rc1_ip_bps1500_slice1_gop30_tu1_fps25.h264 VBR 1500 1026.02 31.00 FAIL
Media_AVC_Enc_VBR/720p5994_parkrun_i420_profilehigh_rc1_ip_bps3000_slice4_gop30_tu4_fps30.h264 VBR 3000 2069.84 31.00 FAIL
Media_AVC_Enc_VBR/720p5994_parkrun_i420_profilehigh_rc1_ipb_bps4000_slice8_gop30_tu7_fps30.h264 VBR 4000 2741.83 31.00 FAIL
Media_AVC_Enc_VBR/720p5994_parkrun_i420_profilebaseline_rc1_ip_bps3000_slice4_gop59_tu4_fps30.h264 VBR 3000 2084.08 30.00 FAIL
Media_AVC_Enc_VBR/720p5994_parkrun_i420_profilebaseline_rc1_ip_bps4000_slice8_gop60_tu7_fps30.h264 VBR 4000 2785.54 30.00 FAIL
Media_AVC_Enc_VBR/1080p_1920x1080_i420_profilemain_rc1_ip_bps2000_slice1_gop30_tu1_fps30.h264 VBR 2000 1512.77 24.00 FAIL
Media_AVC_Enc_VBR/1080p_1920x1080_i420_profilemain_rc1_ip_bps3000_slice4_gop60_tu4_fps30.h264 VBR 3000 2133.23 28.00 FAIL
Media_AVC_Enc_VBR/1080p_1920x1080_i420_profilemain_rc1_ipb_bps4000_slice8_gop59_tu7_fps30.h264 VBR 4000 2887.83 27.00 FAIL
Media_AVC_Enc_VBR/1080p_1920x1080_i420_profilemain_rc1_ip_bps2000_slice1_gop30_tu1_fps25.h264 VBR 2000 1458.21 27.00 FAIL
Media_AVC_Enc_VBR/1080p_1920x1080_i420_profilemain_rc1_ip_bps3000_slice4_gop30_tu4_fps30.h264 VBR 3000 2190.70 26.00 FAIL
Media_AVC_Enc_VBR/1080p_1920x1080_i420_profilemain_rc1_ipb_bps4000_slice8_gop30_tu7_fps30.h264 VBR 4000 2984.89 25.00 FAIL
Media_AVC_Enc_VBR/1080p_1920x1080_i420_profilemain_rc1_ipb_bps2000_slice1_gop30_tu1_fps30.h264 VBR 2000 1507.01 24.00 FAIL
Media_AVC_Enc_VBR/1080p_1920x1080_i420_profilemain_rc1_ip_bps3000_slice1_gop59_tu4_fps30.h264 VBR 3000 2134.75 28.00 FAIL
Media_AVC_Enc_VBR/4K_3840x2160_i420_profilemain_rc1_ip_bps6000_slice1_gop30_tu1_fps30.h264 VBR 6000 3997.46 33.00 FAIL
Media_AVC_Enc_VBR/4K_3840x2160_i420_profilebaseline_rc1_ip_bps7500_slice4_gop30_tu4_fps30.h264 VBR 7500 5094.60 32.00 FAIL
Media_AVC_Enc_VBR/4K_3840x2160_i420_profilehigh_rc1_ipb_bps8000_slice8_gop30_tu7_fps30.h264 VBR 8000 5415.15 32.00 FAIL

wangzj0601, 2018-12-05:
avc vdenc vbr encoding also has this issue.

XinfengZhang, 2018-12-07:
if it is smaller than expectation. it should be not a bug, because VBR should only ensure max bit rate. also, HRD overflow is not an issue.

closed 2019-01-10 by XinfengZhang