intel/media-driver #440: [gstreamer-vaapi][hevc 8bit encode]the bitrate of avc vbr encoded file is out of range target 10%.

wangzj0601, 2018-12-05:
System Environment
Platform: KBL/APL/SKL
Arch: x86_64
Kernel:	4.18.0+
libva	commit 285267586a3d4db0e721d30d4a5f5f9fe6a3c913
media_driver	commit 5976380
gstreamer master branch c45b386b54aea4e707a8be5cbf79dc43623c4614
repo: https://gitlab.freedesktop.org/gstreamer/gstreamer.git
gst-vaapi master branch c728fb4ff6c3b4fc9eb9b43fcea4a22bc15fd678
repo: https://gitlab.freedesktop.org/gstreamer/gstreamer-vaapi.git

Bug detailed description
the bitrate of encoded file is out of range target 10% than predefined setting.

Reproduce Steps
1. build enc as above lists
2. gst-launch-1.0 filesrc location=/media/encoderbitstreams//SOCCER_704x576_30_orig_02.yuv '!' videoparse format=i420 width=704 height=576 framerate=30 '!' videoconvert '!' video/x-raw,format=NV12 '!' vaapih265enc rate-control=vbr bitrate=1500 keyframe-period=30 num-slices=1 max-bframes=0 quality-level=1 '!' video/x-h265,profile=main '!' h265parse '!' filesink location=./SOCCER_704x576_i420_profilemain_rc1_ip_bps1500_slice1_gop30_tu1_fps30.h265
3. bitrate_real=1071.58
4. bitrate_gap=28.00

feiwan1, 2018-12-07:
Hi @wangzj0601 , Please use more details description words and your expection instead of using variable like bitrate_real/bitreate_gap. Thus can make others get a better understanding of what you want to express. :)

XinfengZhang, 2018-12-07:
if it is smaller than expectation. it should be not a bug, because VBR should only ensure max bit rate.  also, HRD overflow is not an issue.

wangzj0601, 2018-12-07:
We hope that the bitrate range can be controlled within target 10%, in this case we set the bitrate as "1500", but it got 1071.58 from the encoded file in fact, the range is out of 10%. In addition, this issue also can be duplicated on hevc 10bit vbr encoding.

wangzj0601, 2019-01-31:
now our test standard has changed for vbr, it become to 0.75*bitrate to 1.1*max-vbv-bitrate, and we add option "max-bframes" into cmdline, then this issue can not duplicated, so close it.
for example
gst-launch-1.0 -vf filesrc location=/media_stream/assets/yuv/SOCCER_704x576_30_orig_02_704x576_I420.yuv num-buffers=150 ! rawvideoparse format=i420 width=704 height=576 framerate=30 ! videoconvert ! video/x-raw,format=NV12 ! vaapih265enc rate-control=vbr bitrate=2142 keyframe-period=30 num-slices=1 max-bframes=0 refs=1 quality-level=1 ! video/x-h265,profile=main ! h265parse ! filesink location=./test.h265

closed 2019-01-31 by wangzj0601