intel/media-driver #441: [gstreamer-vaapi][avc vdenc encode]the encoded file has garbage

wangzj0601, 2018-12-05:
System Environment
Platform: KBL/APL/SKL
Arch: x86_64
Kernel:	4.18.0+
libva	commit 285267586a3d4db0e721d30d4a5f5f9fe6a3c913
media_driver	commit 5976380
gstreamer master branch c45b386b54aea4e707a8be5cbf79dc43623c4614
repo: https://gitlab.freedesktop.org/gstreamer/gstreamer.git
gst-vaapi master branch c728fb4ff6c3b4fc9eb9b43fcea4a22bc15fd678
repo: https://gitlab.freedesktop.org/gstreamer/gstreamer-vaapi.git

Bug detailed description
the encoded file has garbage with iHD driver, but pass with i915 driver

Reproduce Steps
1. build enc as above lists
2. gst-launch-1.0 filesrc location=/media/encoderbitstreams//SOCCER_704x576_30_orig_02.yuv '!' videoparse format=i420 width=704 height=576 framerate=30 '!' videoconvert '!' video/x-raw,format=NV12 '!' vaapih264enc rate-control=cqp init-qp=28 keyframe-period=1 num-slices=4 max-bframes=0 quality-level=4 tune=3 '!' video/x-h264,profile=main '!' h264parse '!' filesink location=./SOCCER_704x576_i420_profilemain_rc0_i_qp28_slice4_gop60_tu4_fps30.h264
3. playback the encoded file

uartie, 2018-12-05:
Duplicate of #389

wangzj0601, 2018-12-06:
close this bug, please refer to #389

closed 2018-12-06 by wangzj0601