intel/media-driver #444: [gst-vaapi][decode] vaGetImage fail for unsupported RT format in DdiMedia_CreateSurfaces2 when using I420 format.

feiwan1, 2018-12-07:
By using gst-vaapi commit 176bbce9, can reproduce this issue.
cmdline:
$ gst-launch-1.0 -vf filesrc location=input.264 ! h264parse ! vaapih264dec ! videoconvert ! video/x-raw,format=I420 ! checksumsink2 file-checksum=false frame-checksum=false plane-checksum=false dump-output=true dump-location=test.yuv

Bug link to gstreamer-vaapi issue:
https://gitlab.freedesktop.org/gstreamer/gstreamer-vaapi/issues/123

feiwan1, 2018-12-07:
Need to add I420 support in DdiMedia_CreateSurfaces2  at:
https://github.com/intel/media-driver/blob/b2cf21234e0ddbea06c87f509462da8d433257bf/media_driver/linux/common/ddi/media_libva.cpp#L1998

XinfengZhang, 2018-12-07:
https://github.com/intel/media-driver/blob/b2cf21234e0ddbea06c87f509462da8d433257bf/media_driver/linux/common/ddi/media_libva.cpp#L2132
 you need to send VASurfaceAttribPixelFormat with correct fourcc, I know that before the VA_RT_FORMAT_YUV420 is I420 by default inside intel-vaapi-driver. but it is NV 12 now

feiwan1, 2018-12-10:
@wangyan42164 , with patch #446 , it still have this issue:

$gst-launch-1.0 filesrc location=a.h264 ! h264parse ! vaapih264dec ! videoconvert ! video/x-raw,format=I420 ! checksumsink2 file-checksum=false frame-checksum=false plane-checksum=false dump-output=true dump-location=test.yuv
Setting pipeline to PAUSED ...
Pipeline is PREROLLING ...
Got context from element 'vaapidecode_h264-0': gst.gl.GLDisplay=context, gst.gl.GLDisplay=(GstGLDisplay)"\(GstGLDisplayX11\)\ gldisplayx11-0";
Got context from element 'vaapidecode_h264-0': gst.vaapi.Display=context, gst.vaapi.Display=(GstVaapiDisplay)"\(GstVaapiDisplayEGL\)\ vaapidisplayegl0";
Redistribute latency...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
0:00:00.073143205 25135      0x1b84990 ERROR       vaapivideomemory gstvaapivideomemory.c:278:map_vaapi_memory: failed to make image current
0:00:00.073168366 25135      0x1b84990 ERROR                default video-frame.c:162:gst_video_frame_map_id: failed to map video frame plane 0
0:00:00.073191930 25135      0x1b84990 ERROR          checksumsink2 gstchecksumsink.c:401:gst_cksum_image_sink_render:<cksumimagesink0> failed to map frame
0:00:00.073288831 25135      0x1b84990 ERROR       vaapivideomemory gstvaapivideomemory.c:278:map_vaapi_memory: failed to make image current
0:00:00.073295272 25135      0x1b84990 ERROR                default video-frame.c:162:gst_video_frame_map_id: failed to map video frame plane 0
0:00:00.073298788 25135      0x1b84990 ERROR          checksumsink2 gstchecksumsink.c:401:gst_cksum_image_sink_render:<cksumimagesink0> failed to map frame
ERROR: from element /GstPipeline:pipeline0/GstH264Parse:h264parse0: Internal data stream error.
Additional debug info:
gstbaseparse.c(3656): gst_base_parse_loop (): /GstPipeline:pipeline0/GstH264Parse:h264parse0:
streaming stopped, reason error (-5)
Execution ended after 0:00:00.003261817
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...

wangyan42164, 2018-12-11:
@feiwan1 So this issue seems to come from gst-vaapi based on Junyan's comments. May you close this issue?

feiwan1, 2018-12-11:
Hi @HeJunyan ， can we close this issue? Is this issue belongs to gst-vaapi?