intel/media-driver #447: [gstreamer-vaapi][avc fei encode]encode fail

wangzj0601, 2018-12-07:
System Environment
Platform: SKL
Arch: x86_64
Kernel:	4.18.0+
libva	commit cf11abe5e1b9c93ee75cf974076957162c1605b9
media_driver	commit ba63dbe1ef0c0ca5f72e01f81ee01178eacae112
gstreamer master branch e501e2f5c9b69dcedaeb89a25a1de29503b1d51f
repo: https://gitlab.freedesktop.org/gstreamer/gstreamer.git
gst-vaapi master branch c728fb4ff6c3b4fc9eb9b43fcea4a22bc15fd678
repo: https://gitlab.freedesktop.org/gstreamer/gstreamer-vaapi.git

Bug detailed description
avc fei enc fail with iHD driver, but pass with i915 driver.

Reproduce Steps

1. build enc as above lists
2. gst-launch-1.0 filesrc location=/media/encoderbitstreams//MOBILE_352x288_30_orig_01.yuv '!' videoparse format=i420 width=352 height=288 framerate=30 '!' videoconvert '!' video/x-raw,format=NV12 '!' vaapih264feienc rate-control=cqp init-qp=7 keyframe-period=30 num-slices=1 max-bframes=0 quality-level=4 search-window=4 len-sp=32 ref-width=8 ref-height=8 submbpart-mask=0x00000010 subpel-mode=1 intrapart-mask=0x00000000 intra-sad=0 multi-predL0=false multi-predL1=true '!' video/x-h264,profile=main '!' h264parse '!' filesink location=./MOBILE_352x288_i420_profilemain_rc0_ip_qp7_slice1_gop30_tu4_feimode1_searchwindow4_lensp32_refwidth8_refheight8_submask0x00000010_subpel1_intramask0x00000000_intrasad0_multiredl0false_multiredl1true_fps30.h264
3. encode fail
error info
Got context from element 'vaapiencodeh264fei0': gst.vaapi.Display=context, gst.vaapi.Display=(GstVaapiDisplay)"\(GstVaapiDisplayX11\)\ vaapidisplayx11-0";
0:00:00.044178995 21189      0x1bc5ad0 ERROR                  vaapi gstvaapiencoder.c:541:gst_vaapi_encoder_put_frame: failed to encode frame (status = -1)
0:00:00.044212665 21189      0x1bc5ad0 ERROR            vaapiencode gstvaapiencode.c:730:gst_vaapiencode_handle_frame: failed to encode frame 0 (status -1)
ERROR: from element /GstPipeline:pipeline0/GstVideoParse:videoparse0/GstRawVideoParse:inner_rawvideoparse: Internal data stream error.

XinfengZhang, 2020-07-04:
@wangzj0601 suppose you are not talking i915 "pass with i915 driver" , and no further support of FEI on gen11, could we close it?

OttoYang, 2023-03-28:
@wangzj0601 could we close this issue? Since FEI is no longer supported on gen11.

wangzj0601, 2023-03-28:
Close it since FEI is no longer supported on gen11.

closed 2023-03-28 by wangzj0601