intel/media-driver #448: [gstreamer-vaapi][vpp]skin-tone-enhancement has no working

wangzj0601, 2018-12-07:
System Environment
Platform: KBL/APL/SKL
Arch: x86_64
Kernel:	4.18.0+
libva	commit cf11abe5e1b9c93ee75cf974076957162c1605b9
media_driver	commit ba63dbe
gstreamer master branch e501e2f5c9b69dcedaeb89a25a1de29503b1d51f
repo: https://gitlab.freedesktop.org/gstreamer/gstreamer.git
gst-vaapi master branch c728fb4ff6c3b4fc9eb9b43fcea4a22bc15fd678
repo: https://gitlab.freedesktop.org/gstreamer/gstreamer-vaapi.git

Bug detailed description
skin-tone-enhancement has no working with iHD driver, but pass with i915 driver.

Reproduce Steps

1. build enc as above lists
2. gst-launch-1.0 -vf filesrc location=/media/gst_vpp/768x480.nv12 '!' rawvideoparse format=nv12 width=768 height=480 '!' videoconvert '!' video/x-raw,format=NV12 '!' vaapipostproc format=i420 width=768 height=480 skin-tone-enhancement=TRUE '!' checksumsink2 file-checksum=false frame-checksum=false plane-checksum=false dump-output=true dump-location=./768x480.nv12_skin-tone-enhancement.yuv
3. gst-launch-1.0 -vf filesrc location=/media/gst_vpp/768x480.nv12 '!' rawvideoparse format=nv12 width=768 height=480 '!' videoconvert '!' video/x-raw,format=NV12 '!' vaapipostproc format=i420 width=768 height=480 skin-tone-enhancement=FALSE '!' checksumsink2 file-checksum=false frame-checksum=false plane-checksum=false dump-output=true dump-location=./768x480.nv12_skin-tone-enhancement_false.yuv
4. md5sum 768x480.nv12_skin-tone-enhancement.yuv
5. md5sum 768x480.nv12_skin-tone-enhancement_false.yuv
6. find md5 value of the two files are same

FurongZhang, 2018-12-29:
I am debugging this issue and will get back to this thread within 2 weeks..

Jister, 2019-06-18:
@wangzj0601 The latest driver has no such issue and can get different results, could you also help to verify it? If the issue is fixed, we can close the issue.

FurongZhang, 2019-06-18:
Good, then, we can close this issue. Thanks Jister.

FurongZhang, 2019-07-15:
@wangzj0601 Can you verify this issue? If no response within 1 week, I want to close this issue since we have verified that the issue is fixed.

wangzj0601, 2019-07-15:
I have a regression test on this issue. It still can be duplicated with the latest iHD driver bc73baa6f1c7cc9fe977aad7caa1885930d867db, thanks

closed 2020-05-14 by Jister