intel/media-driver #452: Duplicate VP shaders (kernels) in media driver

dvrogozh, 2018-12-09:
Looks like there are duplicated shaders (kernels) declared in media driver. At least these:

https://github.com/intel/media-driver/blob/bc71759059b6490d84c2f8547e69116f629527ee/media_driver/agnostic/common/vp/kernel/vpkrnheader.h#L593
https://github.com/intel/media-driver/blob/bc71759059b6490d84c2f8547e69116f629527ee/media_driver/agnostic/common/vp/kernel/vpkrnheader.h#L598

**Question:** which sources are actually published in https://github.com/intel/media-driver/blob/bc71759059b6490d84c2f8547e69116f629527ee/media_driver/agnostic/gen11_icllp/vp/kernel/Source/Save_444Scale16_ARGB_64Byte.asm?

https://github.com/intel/media-driver/blob/bc71759059b6490d84c2f8547e69116f629527ee/media_driver/agnostic/common/vp/kernel/vpkrnheader.h#L594
https://github.com/intel/media-driver/blob/bc71759059b6490d84c2f8547e69116f629527ee/media_driver/agnostic/common/vp/kernel/vpkrnheader.h#L610

**Question:** which sources are actually published in https://github.com/intel/media-driver/blob/bc71759059b6490d84c2f8547e69116f629527ee/media_driver/agnostic/gen11_icllp/vp/kernel/Source/Save_444Scale16_NV21.asm

https://github.com/intel/media-driver/blob/bc71759059b6490d84c2f8547e69116f629527ee/media_driver/agnostic/common/vp/kernel/vpkrnheader.h#L595
https://github.com/intel/media-driver/blob/bc71759059b6490d84c2f8547e69116f629527ee/media_driver/agnostic/common/vp/kernel/vpkrnheader.h#L625

**Question:** which sources are actually published in https://github.com/intel/media-driver/blob/bc71759059b6490d84c2f8547e69116f629527ee/media_driver/agnostic/gen11_icllp/vp/kernel/Source/Save_444Scale16_RGB_64Byte.asm

https://github.com/intel/media-driver/blob/bc71759059b6490d84c2f8547e69116f629527ee/media_driver/agnostic/common/vp/kernel/vpkrnheader.h#L596
https://github.com/intel/media-driver/blob/bc71759059b6490d84c2f8547e69116f629527ee/media_driver/agnostic/common/vp/kernel/vpkrnheader.h#L629

**Question:** which sources are actually published in https://github.com/intel/media-driver/blob/bc71759059b6490d84c2f8547e69116f629527ee/media_driver/agnostic/gen11_icllp/vp/kernel/Source/Save_444Scale16_XRGB_64Byte.asm

The problem is that media driver sources _address_ shaders which do not present in sources (i.e. with upper case SCALE). For example:
```
media_driver/agnostic/gen10/vp/kdll/hal_kernelrules_g10.c:2841:    { RID_SetKernel        , IDR_VP_Save_444SCALE16_ARGB_64Byte , Kdll_None },
media_driver/agnostic/gen11/vp/kdll/hal_kernelrules_g11.c:3620:    { RID_SetKernel        , IDR_VP_Save_444SCALE16_ARGB_64Byte , Kdll_None },
media_driver/agnostic/gen8/vp/kdll/hal_kernelrules_g8.c:2148:    { RID_SetKernel        , IDR_VP_Save_444SCALE16_ARGB_64Byte  , Kdll_None },
media_driver/agnostic/gen9/vp/kdll/hal_kernelrules_g9.c:3604:    { RID_SetKernel        , IDR_VP_Save_444SCALE16_ARGB_64Byte  , Kdll_None },

media_driver/agnostic/gen10/vp/kdll/hal_kernelrules_g10.c:2677:    { RID_SetKernel        , IDR_VP_Save_444SCALE16_RGB_64Byte  , Kdll_None },
media_driver/agnostic/gen10/vp/kdll/hal_kernelrules_g10.c:2707:    { RID_SetKernel        , IDR_VP_Save_444SCALE16_RGB_64Byte  , Kdll_None },
media_driver/agnostic/gen10/vp/kdll/hal_kernelrules_g10.c:2852:    { RID_SetKernel        , IDR_VP_Save_444SCALE16_RGB_64Byte  , Kdll_None },
media_driver/agnostic/gen10/vp/kdll/hal_kernelrules_g10.c:2888:    { RID_SetKernel        , IDR_VP_Save_444SCALE16_RGB_64Byte  , Kdll_None },
media_driver/agnostic/gen11/vp/kdll/hal_kernelrules_g11.c:3448:    { RID_SetKernel        , IDR_VP_Save_444SCALE16_RGB_64Byte  , Kdll_None },
media_driver/agnostic/gen11/vp/kdll/hal_kernelrules_g11.c:3481:    { RID_SetKernel        , IDR_VP_Save_444SCALE16_RGB_64Byte  , Kdll_None },
media_driver/agnostic/gen11/vp/kdll/hal_kernelrules_g11.c:3633:    { RID_SetKernel        , IDR_VP_Save_444SCALE16_RGB_64Byte  , Kdll_None },
media_driver/agnostic/gen11/vp/kdll/hal_kernelrules_g11.c:3671:    { RID_SetKernel        , IDR_VP_Save_444SCALE16_RGB_64Byte  , Kdll_None },
media_driver/agnostic/gen8/vp/kdll/hal_kernelrules_g8.c:1819:    { RID_SetKernel        , IDR_VP_Save_444SCALE16_RGB_64Byte   , Kdll_None },
media_driver/agnostic/gen8/vp/kdll/hal_kernelrules_g8.c:1849:    { RID_SetKernel        , IDR_VP_Save_444SCALE16_RGB_64Byte   , Kdll_None },
media_driver/agnostic/gen8/vp/kdll/hal_kernelrules_g8.c:2159:    { RID_SetKernel        , IDR_VP_Save_444SCALE16_RGB_64Byte   , Kdll_None },
media_driver/agnostic/gen8/vp/kdll/hal_kernelrules_g8.c:2195:    { RID_SetKernel        , IDR_VP_Save_444SCALE16_RGB_64Byte   , Kdll_None },
media_driver/agnostic/gen9/vp/kdll/hal_kernelrules_g9.c:3426:    { RID_SetKernel        , IDR_VP_Save_444SCALE16_RGB_64Byte   , Kdll_None },
media_driver/agnostic/gen9/vp/kdll/hal_kernelrules_g9.c:3456:    { RID_SetKernel        , IDR_VP_Save_444SCALE16_RGB_64Byte   , Kdll_None },
media_driver/agnostic/gen9/vp/kdll/hal_kernelrules_g9.c:3631:    { RID_SetKernel        , IDR_VP_Save_444SCALE16_RGB_64Byte   , Kdll_None },
media_driver/agnostic/gen9/vp/kdll/hal_kernelrules_g9.c:3657:    { RID_SetKernel        , IDR_VP_Save_444SCALE16_RGB_64Byte   , Kdll_None },
```

**Question:** can all these entries, especially in vp gen11+ and vp common code be replaced by addressing shaders for which sources where really published (i.e. Scale)?

zliu50, 2018-12-10:
Open source kernel always uses *Scale* kernel name. if the vpkrnheader.h is online generated , then you should only see *Scale* kernel. *SCALE* kernel might from gen11 previous platform which are not required for kernel open source. it's not duplicated, just different name for different platform.

dvrogozh, 2018-12-10:
1. If _SCALE_ is pre-gen11, then why you use it on Gen11: see https://github.com/intel/media-driver/blob/9b8441382da3d0961030e09aa1017d81fc4fafb0/media_driver/agnostic/gen11/vp/kdll/hal_kernelrules_g11.c#L3447?
2. "just different name for different platform" - that's exactly why it is a duplication. You had a kernel which you have used on some pre-Gen11 platforms. Why you did not use the same name on Gen11? Or, if you decided to refactor something and change the naming convention, why you did not regenerate kernels for pre-gen11 platforms to match?

zliu50, 2018-12-11:
1.  If SCALE is pre-gen11, then why you use it on Gen11: see https://github.com/intel/media-driver/blob/9b8441382da3d0961030e09aa1017d81fc4fafb0/media_driver/agnostic/gen11/vp/kdll/hal_kernelrules_g11.c#L3447?
[Bob] I double check open sourced kernel name, there is no *SCALE* kernel, I would check w/ driver team to fix it in kernel rules.

  1.  "just different name for different platform" - that's exactly why it is a duplication. You had a kernel which you have used on some pre-Gen11 platforms. Why you did not use the same name on Gen11? Or, if you decided to refactor something and change the naming convention, why you did not regenerate kernels for pre-gen11 platforms to match?
[Bob] Kernel for Gen11 previous platform has been existing on different driver branch, and some branch already pv or even freeze to check in, that why we don’t change for gen11 previous platform.

dvrogozh, 2018-12-11:
> [Bob] Kernel for Gen11 previous platform has been existing on different driver branch, and some branch already pv or even freeze to check in, that why we don’t change for gen11 previous platform.

This is a key open source rule. You support all the platforms and code which you have published almost forever.