intel/media-driver #469: How to enable debug/verbose log of media-driver?

enihcam, 2018-12-15:
chromium-vaapi fails to play online video. Currently I can only get logs from chromium, but it is not helping I guess.

```
[1:15:1215/114722.869293:ERROR:render_media_log.cc(30)] MediaEvent: MEDIA_ERROR_LOG_ENTRY {"error":"Format conversion failed."}
[1:15:1215/114722.870482:ERROR:render_media_log.cc(30)] MediaEvent: MEDIA_ERROR_LOG_ENTRY {"error":"Format conversion failed."}
[1:15:1215/114722.871596:ERROR:render_media_log.cc(30)] MediaEvent: MEDIA_ERROR_LOG_ENTRY {"error":"Format conversion failed."}
[1:15:1215/114722.872317:ERROR:render_media_log.cc(30)] MediaEvent: MEDIA_ERROR_LOG_ENTRY {"error":"Format conversion failed."}
[1:15:1215/114722.873042:ERROR:render_media_log.cc(30)] MediaEvent: MEDIA_ERROR_LOG_ENTRY {"error":"Format conversion failed."}
[1:15:1215/114722.874713:ERROR:render_media_log.cc(30)] MediaEvent: MEDIA_ERROR_LOG_ENTRY {"error":"Format conversion failed."}
[1:15:1215/114722.875911:ERROR:render_media_log.cc(30)] MediaEvent: MEDIA_ERROR_LOG_ENTRY {"error":"Format conversion failed."}
[1:15:1215/114722.879423:ERROR:render_media_log.cc(30)] MediaEvent: MEDIA_ERROR_LOG_ENTRY {"error":"Failed to send video packet for decoding: timestamp=2419083 duration=41708 size=2625 side_data_size=0 is_key_frame=0 encrypted=0 discard_padding (us)=(0, 0)"}
[1:15:1215/114722.879981:ERROR:render_media_log.cc(30)] MediaEvent: MEDIA_ERROR_LOG_ENTRY {"error":"Format conversion failed."}
[1:15:1215/114722.880920:ERROR:render_media_log.cc(30)] MediaEvent: MEDIA_ERROR_LOG_ENTRY {"error":"video decoder reinitialization failed"}
[1:15:1215/114722.881219:ERROR:render_media_log.cc(30)] MediaEvent: MEDIA_ERROR_LOG_ENTRY {"error":"Format conversion failed."}
[1:15:1215/114722.881533:ERROR:render_media_log.cc(30)] MediaEvent: MEDIA_ERROR_LOG_ENTRY {"error":"Format conversion failed."}
[1:1:1215/114722.881941:ERROR:render_media_log.cc(30)] MediaEvent: PIPELINE_ERROR PIPELINE_ERROR_DECODE
[2976:2983:1215/114729.391654:ERROR:socket_stream.cc(219)] Closing stream with result -2
[3006:3006:1215/114808.508607:ERROR:vaapi_wrapper.cc(1107)] Failed syncing surface VA error: internal decoding error
[1:15:1215/114808.509896:ERROR:render_media_log.cc(30)] MediaEvent: MEDIA_ERROR_LOG_ENTRY {"error":"video decode error"}
[1:1:1215/114808.513472:ERROR:render_media_log.cc(30)] MediaEvent: PIPELINE_ERROR PIPELINE_ERROR_DECODE

```

How to enable debug/verbose log of media-driver?

eero-t, 2019-01-16:
> It looks like this issue only happens to WEBM format.
> MP4 format is good.

Both of these are containers formats, what's the actual failing codec, VP8?

(e.g. mediainfo program in Linux would give more info on the video format.)

Does "dragging progress bar" mean that failure happens at some specific position in video?

Can you reproduce the issue e.g. with FFmpeg or sample applications in MediaSDK?

enihcam, 2019-01-17:
Repo steps:
1. boot with Manjaro LiveCD:
2. yaourt -S intel-media-driver chromium-vaapi (or chromium-vaapi-bin)
3. change vaapi driver to iHD
4. enable hardware video acceleration in chromium about:flags
5. Try "dragging progress bar" for the the following videos:
https://www.billboard.com/video/janet-jackson-meets-bts-backstage-at-mnet-asian-music-awards-billboard-news-8490370
https://www.nhl.com/video/icymi-lundqvists-stellar-save/t-277350912/c-64029403

widefox, 2019-01-25:
I don't know about WEBM but MP4 fails e.g. mp4, 1920x1080, avc1.640028, 24fps. Same issue on up to date Fedora 29, kernel 5.0rc3, Chromium-vaapi Version 71.0.3578.98 (RPMFusion Build) chromium-vaapi (64-bit), iHD, HW accel enabled, ffmpeg v4.0.3 . Does not happen with unaccelerated Chrome 71, Chrome beta 72. YouTube videos H264 (to reproduce use h264ify extension). Default WEBM YouTube with VP9 (1920x1080, 2754k , vp9, 24fps,) is not hardware accelerated on this PC and does not have this bug.

maximbaz, 2019-02-10:
VA-API patch is now part of official Chromium on Arch Linux, so be aware that `chromium-vaapi` and `chromium-vaapi-bin` are obsolete now, also HW video acceleration is enabled by default so no need to mess with Chromium flags to get it enabled. 

This also means that the bug is suddenly more important because potentially more people are affected by this now 😉 

I can confirm that the issue can only be reproduced with H264 format, VP9 is working fine. I used `youtube-dl` to download a random youtube video in two formats, `(h264 1920x1080 60.000fps)` and `(vp9 1920x1080 60.000fps)`. Open both in Chromium from the local file system, then try to move forward in the video — H264 video will get stuck, and you won't be able to play it anymore, while VP9 video correctly continues playing from the new position.

Maybe a simpler way to reproduce the issue is to install [h264ify](https://chrome.google.com/webstore/detail/h264ify/aleakchihdccplidncghkekgioiakgal) extension, then just navigate to YouTube and play any video, click on "progress bar" to move further and video will get stuck.

This is what is printed in console at this time:

```
[18209:18209:0210/215931.479441:ERROR:vaapi_wrapper.cc(1149)] Failed syncing surface VA error: internal decoding error
[103:175:0210/215931.479643:ERROR:render_media_log.cc(30)] MediaEvent: MEDIA_ERROR_LOG_ENTRY {"error":"VDA Error 4"}
[103:175:0210/215931.479788:ERROR:render_media_log.cc(30)] MediaEvent: MEDIA_ERROR_LOG_ENTRY {"error":"video decode error"}
[103:103:0210/215931.480022:ERROR:render_media_log.cc(30)] MediaEvent: PIPELINE_ERROR PIPELINE_ERROR_DECODE
```

widefox, 2019-02-17:
Confirm H264 seeking still broken in chromium-vaapi.x86_64                                                                         72.0.3626.81-2.fc29 libva-2.3.0-1.fc29.x86_64 intel-media-driver.x86_64                                                                         18.3.0-2.fc29  . chromium-vaapi is the package name in Fedora from rpmfusion (rpmfusion-free-updates). Note that currently F29 now needs downgrade of libva from 2.4 to 2.3  libva-2.4.0-2.fc29 -> libva-2.3.0-1.fc29 to enable acceleration, else will get warning "[4138:4138:0217/130657.111905:ERROR:vaapi_wrapper.cc(343)] This build of Chromium requires VA-API version 1.3, system version: 1.4"

enihcam, 2019-02-22:
Hi Intel, when will you start debugging on this?
latest chromium + latest media-driver does not resolve the issue.

widefox, 2019-02-27:
Appears fixed on Fedora 29 (rpmfusion testing) chromium-vaapi 72.0.3626.109 (RPMFusion Build) chromium-vaapi (64-bit), libva 1.4 . [enihcam](/enihcam) do you have details?

maximbaz, 2019-02-27:
Are you sure @widefox? I don't think there have been any releases in this repo, and Fedora's spec doesn't seem to use any new patches. Maybe you are accidentally using a different driver?

widefox, 2019-02-28:
> Are you sure @widefox? I don't think there have been any releases in this repo, and Fedora's spec doesn't seem to use any new patches. Maybe you are accidentally using a different driver?

@maximbaz Yes. Tested on 2 Intel i5/i7 laptops (5th gen and 2nd gen). Confirm no seeking problem now. I may have the hybrid driver installed now too when I didn't before I don't know, as I varied many things when I got the libva 1.3 and 1.4 version mismatch for a week which prevented hardware acceleration (as I reported above - which I presume the package maintainer should fix the chromium-vaapi package dependency so the matching 1.3 wouldn't be updated to an incompatible 1.4).

F29 updated from libva 1.3 -> 1.4 a weekish ago. That resulted in a version incompatibility with the then latest chromium-vaapi from RPMFusion and AFAIK the chromium from Fedora (as I reported above). Now the new RPMFusion chromium-vaapi 72.0.3626.109 is compiled with 1.4 (rather than 1.3 previously) so the incompatibility is gone, and CPU with fullscreen YouTube goes from roughly 170% to 15-30% with H264 1080p 30Hz or 60Hz . I can confirm seeking does not crash now. CAVEAT: 1. the browser is crashing sometimes now when it didn't before but that may due to 2. changes to my about:flags - I haven't isolated yet.

hexchain, 2019-02-28:
The old libva-intel-driver may be in use if it's installed. Could you please verify that you are really using the new media-driver? e.g. run applications (vainfo, chromium) with environment `LIBVA_DRIVER_NAME=iHD` and try to reproduce the problem.

widefox, 2019-02-28:
@hexchain @maximbaz my mistake - you're both right. It still fails with the iHD driver - 1.0.0 . It works with the Intel i965 driver. I must have only had the iHD driver before the 1.3/1.4 issue, but now have both. (This is with ffmpeg version 4.0.3 )

enihcam, 2019-02-28:
@maximbaz @hexchain can you find more details from driver logs instead of chromium logs, and then open a new bug? I don't think they will take a look at this 'chromium issue'. You know what i mean.

By the way, i have switched back to i965 driver already, because I'll get an AMD laptop next month.

maximbaz, 2019-03-01:
unfortunately I have no idea how to do that - could you please help the issue open? many people have already subscribed to it

enihcam, 2019-03-01:
https://01.org/linuxgraphics/documentation/bugs-and-debugging

enihcam, 2019-03-05:
Try replace ffmpeg with ffmpeg-qsv?

Spongman, 2019-04-25:
how do you set an environment variable for a program that's running inside a snap?

wangyan-intel, 2019-08-30:
It should be fixed by https://github.com/intel/media-driver/pull/562. This PR have been merged. Keep this issue one week. If no more response, I will close it.

maximbaz, 2019-08-30:
I can confirm that I no longer reproduce the issue, great work!

wangyan-intel, 2019-09-01:
@maximbaz Thanks for your confirmation.

closed 2019-09-01 by wangyan-intel