intel/media-driver #471: shaders: warning during VP shaders generation

dvrogozh, 2018-12-17:
Apply #453, you will see the following warning during shaders generation:
```
[ 63%] Generating return.hex
line 24.10: warning: destination register ARF access requires {Switch} ThreadCtrl
         mov    (1|M0)    ip.0<1>:ud    r8.7<0;1,0>:d
         ^
```

It is coming from here:
https://github.com/intel/media-driver/blob/273ef13413a1c1f3527831df9f2ad962361dd97c/media_driver/agnostic/gen11_icllp/vp/kernel/Source/return.asm#L24

I've been on this commit in IGC (https://github.com/intel/intel-graphics-compiler):
```
commit 4128b9e13ed44ebce29a8df7fbeea4b7a5486afd
Author: Wei Pan <wei.pan@intel.com>
Date:   Thu Nov 8 18:25:27 2018 +0100

    Limit clustering width

    Change-Id: I9ee7beed0689553d7ffe0826a68950d6462f6417

```

FurongZhang, 2018-12-29:
Will check that.

FurongZhang, 2020-05-26:
Kernel has a fix.  Requesting to have a try.

FurongZhang, 2020-06-02:
@XinfengZhang , We sent the binary to bug submitter via Email, not get any response. Need you to check with the bug submitter.

XinfengZhang, 2020-07-01:
I believe there should be a PR to fix it. @FurongZhang

FurongZhang, 2021-04-26:
I have requested shader folks to submit a PR to fix this warning.

closed 2021-12-21 by FurongZhang